shaise/FreeCAD_SheetMetal #330: FOLD A WALL              Position = "intersection of planes"

free777cat, 2024-04-03:
Unfortunately, I don't have free time now. But it works well!
[Uploading SheetMetalFoldCmd.py___v.2.zip…]()

[image]
[image]
[image]

free777cat, 2024-04-03:
adaptive          k-factor

free777cat, 2024-04-13:
Maybe add it?

shaise, 2024-04-13:
I'm not sure I understand what you want.

pierreporte, 2024-04-13:
Sounds like #307.

free777cat, 2024-04-14:
[https://youtu.be/AIOcVb3u4G0](url)
example, my code works

free777cat, 2024-04-14:
[SheetMetalFoldCmd.py.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/14854941/SheetMetalFoldCmd.py.zip)

shaise, 2024-04-18:
OK i understand now. Thanks.
I will do some tests.

shaise, 2024-04-18:
I checked your code.
Its not working.
First of all: Forward, middle, backward does not work any more!!!
2. intersection of planes of outside fold does not work.
Inside works: 
[image]

Outside (like issue 307) not working:
[image]

free777cat, 2024-04-20:
There was indeed an error that prevented other functions from working.
Fixed!!!
Everything works fine now.
In the code, you still need to return the "adaptive k-factor" value to the property field.

[SheetMetalFoldCmd.py.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/15047213/SheetMetalFoldCmd.py.zip)

[https://youtu.be/N0BfopASosw](url)

shaise, 2024-04-20:
Yes!
Works very good! thanks!
Some questions:
1. You changed the default kfactor from 0.5 to 1. Do you must? this can hurt compatability with old models.
2. you use adaptable k factor. this means that I can not use manual k factor?

shai

free777cat, 2024-04-20:
Q1 - No.That was the development process.
Q2 - When designing, it is more important to preserve linear dimensions.
    In my "intersection of planes" code, the effect of manually entering the kfactor is excluded.
    Note the printout of the kfactor in the report (below) when this code is running.
    I think adaptive k-factor should be used everywhere. (The possibility of manual entry as an additional option).
    This is the "way" to collapse-expand (with size preservation) without errors.

    In fact, the influence of the "manual k-factor" is important at the stage of final deployment of the finished model to obtain the "cutting drawing".
    It is necessary here!!!

free777cat, 2024-04-20:
Please note that when designing the "k-factor" take the unknown in advance.

shaise, 2024-04-20:
The problem is that in real world the k factor is depending only in the material, not on the model. When you unfold the part, you must use the correct k-factor, not the adaptive one. this means that when you fold a wall you need to set the wall length based on the K factor, not set the k factor by trying to preserve correct lengths.
Another option is to not use K-Factor at all when folding, but then when you unfold you can get different size then original.

free777cat, 2024-04-20:
In addition to the material, the "k factor" can depend on the tool and the processing method.
The output data are exactly the dimensions that we have to get.

If, for example, there are two different grades of steel in production.
This does not affect the construction and dimensions of the model.
This affects the calculation of various workpiece drawings based on the model.

shaise, 2024-04-20:
Ok as long as the other options stay the same, its ok with me.
Do you want to create a pull request or do you want me to push the changes?

free777cat, 2024-04-20:
Make changes.
Could you also add that the adaptive k-factor (calculated) is displayed in the field?

shaise, 2024-04-20:
Again, I don't understand what is the usage of changing. We can not "invent" K factor, if user uses material with k-factor 0.7, and when you make fold he sees k factor 0.4 what he will do with it?

shaise, 2024-04-20:
I think k-factor should determine what will be the end wall height, based on the material.

free777cat, 2024-04-20:
I suggest making "intersection of planes" the first function. (Default)
We make the "k factor" in this function adaptive without the possibility of change.
If the user needs to change it, he will switch to other options.

free777cat, 2024-04-20:
> I think k-factor should determine what will be the end wall height, based on the material.

When designing, the user may not know the "k factor" of the specific material that will be used in manufacturing.
The user needs to get a model according to the data known to him: height, thickness, length.

free777cat, 2024-04-20:
In addition, "k factor" is a coefficient that cannot be measured instrumentally.

shaise, 2024-04-20:
I completely agree with anything you say.
The fold wall does not need to use K factor at all when you design a new part. Only when you unfold it you need to use k factor.
However there are 2 counter points,
1. If I change it for the old options, it will not be backward compatible.
2. Sometimes a user get a drawing of a flat sheet metal that was flatten with k factor, and he wants to see how it will look like after folding. (not designing a new part, just to 'reverse engineer' an existing drawing.

I see 2 possible solution:
1. we use your code as-is. old options are backward compatible, new option is  'adaptive k'. but we dont update the fiels as it have no real usage. Its like not using k factor at all.
2. better solution: add a new boolean field: UseKFactor if it is true - use the k factor in calculations. if not, use without. it will be true by default for old options, and false for 'intersection planes', but anyone can change it.

free777cat, 2024-04-21:
> 2\. Sometimes a user get a drawing of a flat sheet metal that was flatten with k factor, and he wants to see how it will look like after folding. (not designing a new part, just to 'reverse engineer' an existing drawing.

I would like to note that it is almost impossible to do this with the existing "old" collapsing functions.
It is for this reason that I began to develop the "intersection of planes"

free777cat, 2024-04-21:
> 2\. better solution: add a new boolean field: UseKFactor if it is true - use the k factor in calculations. if not, use without. it will be true by default for old options, and false for 'intersection planes', but anyone can change it.

A very good decision.

A radical solution would be to create two new elements: "Fold a wall only adaptive" and "make a wall only adaptive", but this will probably complicate the work process in the future.
I don't see a way to change for example: "Fold a wall only adaptive" to "Fold a wall" in the existing document.

shaise, 2024-04-21:
OK let me take a look and see what I can do.

free777cat, 2024-04-21:
> A radical solution would be to create two new elements: "Fold a wall only adaptive" and "make a wall only adaptive", but this will probably complicate the work process in the future. I don't see a way to change for example: "Fold a wall only adaptive" to "Fold a wall" in the existing document.

I didn't quite understand what you think about this?

shaise, 2024-04-21:
definitely not the radical solution.
For now I may use your code as-is, no changes, and maybe add the "useFactor" parameter.

free777cat, 2024-04-21:
Ok, let's move in this direction. Do you need further explanation about my code?

free777cat, 2024-04-21:
And another question: could we integrate a "bend" here through the intersection of the selected plane of the model and, for example, the plane from "SubShapeBinder"?

closed 2024-04-28 by free777cat