shaise/FreeCAD_SheetMetal #331: Update for new Material System

maxwxyz, 2024-04-07:
With the merge of @davesrocketshop Material overhaul, the Sheet Metal features seem not working as they are missing a shape color attribute:
[image]
 
```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36784 (Git)
Build type: Release
Branch: main
Hash: d1930fc7e6fc35c413f9f39d22f2ea28fca827c7
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.10
  * CurvedShapes 1.0.8
  * Curves 0.6.29
  * Defeaturing 1.2.2
  * fasteners 0.5.16
  * FEMbyGEN 2.1.0
  * freecad.gears 1.2.0
  * freecad_metal_workbench 0.0.1
  * kicadStepUpMod 11.1.0
  * lattice2 1.0.0
  * OpenDark 2023.12.17
  * Rocket 3.3.0
  * sheetmetal 0.4.10
  * Silk 0.1.5

```

davesrocketshop, 2024-04-07:
ShapeColor is implemented as a custom attribute. There should be no issue. But of course that doesn't mean there isn't.

Please include simple steps to reproduce as I'm not familiar with the workbench

maxwxyz, 2024-04-07:
I'm not sure if related to Material but it worked with the last weekly build before the Material merge.
Just select any edge or point and try any commands, no feature is working.
You can create a base shape with the right button and select another edge and click e.g. bend: 
[image]

shaise, 2024-04-07:
if shapeColor no longer available it will break Fasteners as well.
Isn't it backward compatible?

maxwxyz, 2024-04-07:
With a quick test, Fasteners WB works as usual. Maybe there was another change introduced?

shaise, 2024-04-07:
@maxwxyz Do you have the latest version?
In the latest version (as of yesterday) I added an option to set custom default color for fasteners in the preferences.

maxwxyz, 2024-04-07:
@shaise you are right, I was using 0.5.15.
But also adding or moving fasterners with the 0.5.16 works fine.

shaise, 2024-04-07:
I understand why it has no effect. it just sets the diffuse color attrib, but this attrib has no meaning anymore. This is not good.
It should be backward compatible.
Regarding sheetmetal, I will try a new approach and let you know.

shaise, 2024-04-07:
Hi @maxwxyz ,
The feature that was broken is the material preservation system. This means that if you change the color of a sheetmetal part, each time you add a new bend or other sheetmetal feature, the color will be preserved and not revert to gray. 
I have changed the preservation system to be more robust and it now works both on new and old builds.
Please try and let me know.
However what I don't like is that in the new builds there is no way to change the color of a part without setting a new material. Even setting "Random Color" which I use a lot has no effect!
This is probably why fasteners default color no longer works.

shai

maxwxyz, 2024-04-07:
The SM features work now! But I cannot set a Material or ShapeColor to the body

maxwxyz, 2024-04-07:
But this seems to be the issue with https://github.com/FreeCAD/FreeCAD/issues/13344 I can apply a Material to a specific SM feature, e.g. a Bend