shaise/FreeCAD_SheetMetal #334: object has no attribute 'Object'

free777cat, 2024-04-27:
Traceback (most recent call last):
  File "/home/USER/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalBaseCmd.py", line 295, in claimChildren
    if hasattr(self.Object, "BendSketch"):
               ^^^^^^^^^^^
<class 'AttributeError'>: 'SMBaseViewProvider' object has no attribute 'Object'

[Object.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/15137416/Object.zip)
[image]

free777cat, 2024-04-27:
[image]

shaise, 2024-04-27:
Thanks. I'm checking this

shaise, 2024-04-27:
Should be fixed now.

free777cat, 2024-04-27:
Дякую !!!

closed 2024-04-28 by free777cat