shaise/FreeCAD_SheetMetal #336: Problem adding italian translation

robbeban, 2024-05-12:
Hallo,
i want add italian translation to SheetMetal addon.
I followed the instructions but I can't find update.sh file in translations directory.
How can I do?
TIA and thanks for your work

shaise, 2024-05-13:
I think the file you are looking for is [update_translation.sh](https://github.com/shaise/FreeCAD_SheetMetal/blob/master/translations/update_translation.sh). It resides in the translation directory.

luzpaz, 2024-09-09:
@robbeban bump

hasecilu, 2024-09-25:
It's almost finished on CrowdIn :+1: ; there is a big open PR here :tada:, after it's merged I expect some changes on strings so we would need to more time to finish translations.

luzpaz, 2024-12-04:
@hasecilu this still relevant ?

hasecilu, 2024-12-04:
Italian is at 74% works fine.

closed 2024-12-04 by luzpaz