shaise/FreeCAD_SheetMetal #341: Fold a Wall ===>   Redundant "icons" enabled and working

free777cat, 2024-05-29:
[image]
[image]
[image]
[image]

free777cat, 2024-05-29:
When I was working quickly in the morning and made a mistake with the button, I almost fell (in a stupor)

shaise, 2024-05-29:
Thanks, I will take a look at it.

shaise, 2024-05-29:
@free777cat , should be ok now.

free777cat, 2024-05-29:
Ok !!!!!!!!!!!!!
Thanks.

closed 2024-05-29 by free777cat