shaise/FreeCAD_SheetMetal #343: Errors when recompute files created in versions 0.20 - 0.21

GS90, 2024-06-08:
Hello,
When recalculating or trying to change the geometry, the following error occurs:

```
16:00:32  pyException: Traceback (most recent call last):
  File "/home/***/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalCmd.py", line 1584, in execute
    s, f = smBend(
           ^^^^^^^
  File "/home/***/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalCmd.py", line 955, in smBend
    mainlist, trimedgelist, nogaptrimedgelist = getBendetail(
                                                ^^^^^^^^^^^^^
  File "/home/***/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalCmd.py", line 507, in getBendetail
    lenEdge, selFace, thk, revAxisV, thkDir = smEdge(selItemName, MainObject)
                                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/***/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalCmd.py", line 492, in smEdge
    thk, thkDir = sheet_thk(MainObject, selFaceName)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/***/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalCmd.py", line 460, in sheet_thk
    thkDir.normalize()
    ^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'NoneType' object has no attribute 'normalize'
16:00:32  Bend001: 'NoneType' object has no attribute 'normalize'
```

This error occurs only in files created in versions prior to the TNP fix.
Saving the file in the latest version does not help.

https://github.com/shaise/FreeCAD_SheetMetal/assets/25671299/4796328e-57cd-419c-b536-111665973799

[example.zip](https://github.com/user-attachments/files/15748158/example.zip)

FreeCAD info:

```
OS: Ubuntu 22.04.4 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37668 (Git) AppImage
Build type: Release
Branch: main
Hash: ea68c5e88cdc80044ede482f6ad908820f4c1c82
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.13
  * Plot 2022.4.17
  * sheetmetal 0.4.18
  * addFC 0.3.1
  * CfdOF 1.25.17
  * fasteners 0.5.21
```

shaise, 2024-06-08:
Hi @GS90 ,
Thanks! I was able to reproduce it. This again seems as a TNP issue, but this time its in a level I can not control through the WB. 
This happens also on new parts created by latest freecad. 
I'm waiting for a TNP fix for the previous issue, and then I will see if it fixes this issue as well.

luzpaz, 2024-09-10:
Related to #342