shaise/FreeCAD_SheetMetal #344: Can we have properties converted to Title Case to match Freecad style?

leoheck, 2024-06-20:
Hey @shaise nice to meet you here. I did not know that this was also your work. 
What do you think about updating property names on this Workbench too?

[image]

shaise, 2024-06-20:
Yes, I had a feeling it will come into this. Sheetmetal is a bit more complicated in that matter. But I will take a look as well.

leoheck, 2024-06-20:
> Yes, I had a feeling it will come into this.

Sorry to give you this much work dude!

luzpaz, 2024-09-10:
I guess there would need to be a slow deprecation process whereby the deprecated properties need to be converted by a script for backward compatibility, notify the user that older versions of SheetMetal addon aren't compatible... ?

leoheck, 2024-09-10:
What If this case insensitive when Reading from the file.. and converted to title case and when saved replaced the title case in use?

shaise, 2024-09-10:
I think that I will probably end up doing the same as I did in Fasteners. Only reason I postponed it is because I need to see how I merge the changes that were done for Ondsel's sheetmetel.

luzpaz, 2024-12-04:
@shaise is this still relevant ?

leoheck, 2024-12-04:
> UI Task Panel
Which UI Task Panel?

shaise, 2024-12-04:
with the new version, there is a task panel for this function

leoheck, 2024-12-04:
The new version of Freecad or the Workbench? So no properties on the properties panel anymore?

shaise, 2024-12-04:
new version of sheetmetal:
[image]

leoheck, 2024-12-04:
Nice, and those settings won't be visible on the properties panel?

shaise, 2024-12-04:
the property panel is the low level of any feature and will always work. The UI is only a helper.

leoheck, 2024-12-04:
But can we have the title case on the property panel as well to mimic other properties? This will give the professional look that Freecad deserves.

shaise, 2024-12-05:
@luzpaz , what do you think?