shaise/FreeCAD_SheetMetal #345: Refine property missing.

leoheck, 2024-06-20:
Where is this Refine property, I could not find it. Is this generic advice or this feature should have this property and I am not finding it?
[image]

leoheck, 2024-06-20:
I am using 
```
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37819 (Git) AppImage
  * sheetmetal 0.4.18
```

Trying to unfold a simple structure.. but it is failing. Actually this is the first time I am doing this, so I am not sure if I am doing this right.

I am selecting a face and trying to unfold the structure.

The structure is simple, I guess. And I am also seeing this issue here:

[image]

```
23:11:35  Error Analysis: Sheet thickness invalid for this face! at Face39
23:11:35  exception at line 3131("'NoneType' object has no attribute 'Edges'",)
23:11:35  ("'NoneType' object has no attribute 'Edges'",)
```

leoheck, 2024-06-20:
There is also a different issue when I apply Part Design Features. I am not sure if it supports that.

```
23:31:40  Error Unfold: not handled curve type in unbendFace at Face146
23:31:40  exception at line 3131("'NoneType' object has no attribute 'Edges'",)
23:31:40  ("'NoneType' object has no attribute 'Edges'",)
```

For instance, I tried to make a groove in the last feature to create some room inside.

[image]

[image]

shaise, 2024-06-20:
the refine is a general preference in part design.
please upload the file that failed unfold here.

leoheck, 2024-06-20:
The first folder has the one that fails for some unknown reason. This was my first attempt, and things broke during the process and I had to fix some features after some updates adding extra faces and this may be related to the issue. I am about to redo this to check if I can improve it. I was doing features on one side, and after I realized that I could select more than 1 face at the same time so I don't have to mirror anything at the end. 

The second folder has another similar test. To see if a new object could work without issues. And it worked fine including the Pocket.

The 3rd folder is another test, where I am using a Grove as in the 1st test, and it works, but the projected Sketch is not as good as in the second test.

[unfold_test.zip](https://github.com/user-attachments/files/15914958/unfold_test.zip)

leoheck, 2024-06-20:
> the refine is a general preference in part design.
please upload the file that failed unfold here.

The popup suggested to turn off refine on Bend02. 

Ah, I understand now that the suggestion was to disable the refine somewhere until the Bend02 feature since Bend02 does not have that property.

leoheck, 2024-06-20:
Replicated the design without the groove, and it works.

Also, I could not set the tip of the Body with the Sheetmetal, to ignore the Part Design Feature, I had to delete it. Is it possible to do that somehow? This way I could test intermediate places to check if the sheet metal is working.

shaise, 2024-06-22:
Few things about the original design:
1. Bend 4 had a bend radius of 0. This can tend to cause issues.
2. The groove cut into the some bends and created  'refined' faces crossing both flat and curved features. This tends to confuse the unfolder.
3. the 'switch refile to false for feature xxx' usually does not correspond to any particular bend on the way. Its just the name of the body you have selected to unfold. You can export your part as step, import to a different document and get the same sort of error. The unfolder is not 'aware' of any folds. It works on the selected part as simple solid object.

leoheck, 2024-06-23:
Good. I don't remember setting something with radius zero. At least least this was not my intention. About the groove, am I doing something wrong or is there any design trick that I can to do avoid confusing the unfoder that much?

shaise, 2024-06-23:
There is no right or wrong when cutting features into a sheetmetal object. Its just a limitation of the unfolder (the unfold algorithm is very complex and was done by a very clever programmer - not me)
One thing to take into account though, is that some of the groove cuts create edges in the sheet that are not completely perpendicular to the sheet surface, hence even if you unfold it correctly, the resultant sheet pattern can not be cut using a laser as the laser creates only perpendicular edges.

leoheck, 2024-06-23:
Thanks for the explanation. I was experimenting this workbench to design a cardboard structure to fix items inside a cardboard box.

luzpaz, 2024-09-10:
Hard to follow due to multiple issues reported. What is relevant here?

leoheck, 2024-09-10:
The issue is that a bad message appears when something doesn't work. 

The message referes to a Metal sheet feature that doesn't have the propert.y Refine.

PR #433 (2025-01-06): add new unfolder backend
This PR is to make sure we don't forget to add the new dependency in `package.xml` and close relevant issues. @shaise will plug in the backend functions to the GUI as needed.

This new unfolder:
- Uses the `networkx` python module for analyzing the sheet faces
- Should be indifferent to refined vs. unrefined shapes. Seam lines don't matter in most cases
- Is **5-50x faster than the old unfolder**  

closes #408
closes #361 
closes #345
closes #284 
closes #279 
closes #278 
closes #274 
closes #272 
closes #254 
closes #235 
closes #225 
closes #203 
closes #25 
closes [... cut]
closed 2025-01-06 by shaise