shaise/FreeCAD_SheetMetal #346: Base Feature images do not match the color theme of the icons

leoheck, 2024-06-20:
Sheet metal toolbar icons are awesome. 
I just realized that these icons to select the origin of the Base Feature do not match the theme. 

What do you think about updating the colors of those images using blue instead?

I would also suggest changing the origin dot/ball with something like and + with arrows but this can be too much, maybe.

[image]

leoheck, 2024-06-27:
@shaise I am seeing this new way of selecting the origin, it looks nice, but the image looks a bit wrong. 
Is this as you expected?

[image]

shaise, 2024-06-27:
it is wrong. should look like this:
[image]

shaise, 2024-06-27:
This is rather strange as in your version the icons exceeds the maximum with which is 28 pixels

leoheck, 2024-06-27:
I am going to check on discord if this is due to the OpenDark Theme.

leoheck, 2024-06-27:
Maybe the min size property is missing somewhere

shaise, 2024-06-27:
its not an opendark issue
[image]

shaise, 2024-06-27:
i have updated minimum size as well. please try it (you might need to uninstall/reinstall since i did not update the version)

leoheck, 2024-06-27:
Still the same, and there are these pushXX messages, maybe from this wb.
[image]

shaise, 2024-06-27:
Another possible tweak. Can you try again? (also removed pushXX)

closed 2024-06-27 by shaise

shaise, 2024-06-27:
If it does not work, I will try to get a ubuntu machine.

leoheck, 2024-06-27:
Prints were removed but the image/button is still the same.
[image]

shaise, 2024-06-28:
I have set the policy to fixed. can you try again?

leoheck, 2024-06-28:
sure! let me check.

leoheck, 2024-06-28:
[image]

It is still the same. 
[image]

Perhaps it is a problem with the `AppImage` package.

shaise, 2024-06-29:
Ok I fired up my ubuntu.
on build 37574, standard theme it looks like this:
[image]
On opendark:
[image]

So it is definitely an opendark issue but only on linux. Can you switch back to standard theme and check to be sure it is the issue?

shaise, 2024-06-29:
another update:
Using openTheme (dark) instead of openDark seems to solves the issue (need to uninstall openDark)
[image]

leoheck, 2024-07-02:
This took me time. Yeah, I confirm the light version works. 
[image]

So the openTheme is the most updated,... this is kind of hard to track. haha thanks.
Thanks for your patience.

commit 151ce5b6 (2024-06-27): remove debug print and another try to fix #346