shaise/FreeCAD_SheetMetal #348: Base shape task panel is too tall

pierreporte, 2024-06-23:
The task panel to create a base shape is nearly twice taller than needed:

[image]

shaise, 2024-06-23:
The size is determined by freecad. In my case it looks like this:
[image]

pierreporte, 2024-06-23:
It’s strange! I’m running this:

```
OS: Ubuntu 23.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37841 (Git) AppImage
Build type: Release
Branch: main
Hash: 5f761c758e2a0164b00a7183b9c0cbe32688e123
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * CosmeticThread3D
  * Curves 0.6.39
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.19
```

shaise, 2024-06-23:
As far as I understand It will automatically span as low as the screen allow. I will check if I can do it differently

luzpaz, 2024-09-10:
@shaise  any follow-up ?

shaise, 2024-09-10:
@luzpaz , not sure what to do here, the task dialog does not impose size restrictions