shaise/FreeCAD_SheetMetal #351: Bridge fails to execute

gilmiz, 2024-06-27:
After selecting faces as described in the [instructions](https://wiki.freecad.org/SheetMetal_Forming), the sheet to be embossed disappear with the following error message:

`15:55:17  PartDesign::FeaturePython: Link(s) to object(s) 'Pad001' go out of the allowed scope 'WallForming'. Instead, the linked object(s) reside within 'Part Part'.
15:55:17  pyException: Traceback (most recent call last):
  File "C:\Users\gmi\AppData\Roaming\FreeCAD\Mod\sheetmetal\.\SheetMetalFormingCmd.py", line 197, in execute
    a = makeforming(tool, base, base_face, thk, tool_faces, offsetlist[i], fp.angle.Value)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\gmi\AppData\Roaming\FreeCAD\Mod\sheetmetal\.\SheetMetalFormingCmd.py", line 141, in makeforming
    offsetshell_tran = transform_tool(offsetshell, base_face, tool_faces[0], point, angle)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\gmi\AppData\Roaming\FreeCAD\Mod\sheetmetal\.\SheetMetalFormingCmd.py", line 125, in transform_tool
    tool.rotate(rot_center, rot_axis, -rot_angle)
<class 'Part.OCCError'>: class Standard_ConstructionError gp_Dir() - input vector has zero norm
15:55:17  WallForming: class Standard_ConstructionError gp_Dir() - input vector has zero norm`

FC details:
```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Danish/Denmark (da_DK)
Installed mods: 
  * Assembly4 0.50.14
  * BillOfMaterials 0.0.8.5
  * dodo 1.0.1
  * fasteners 0.5.24
  * Render 2024.6.5
  * sheetmetal 0.4.20

```

test file attached:
[test.zip](https://github.com/user-attachments/files/16015095/test.zip)

shaise, 2024-06-30:
Strange. On freecad 0.21 it works well, I'm not sure yet what changes caused it to stop working on 0.22
[image]

shaise, 2024-06-30:
Hey @gilmiz ,
I think I found the problem. Please update and try again.
[image]

gilmiz, 2024-06-30:
Hi @shaise ,
Thanks. 
The fix seems to work. I will test some more tomorrow.

gilmiz, 2024-07-01:
As far as a I can see, its working correct.

closed 2024-07-01 by gilmiz