shaise/FreeCAD_SheetMetal #353: Cleaning a field result in a temporary error

leoheck, 2024-07-07:
Since the Base Feature is updated every time the user updates the input field, cleaning them results in an error.
I guess, Freecad does a bit differently when updating fields. It waits until the focus leaves from the field. I wouldn't say I like this much but it is better than throwing an error on the Report view. However, sheetmetal could wait until there is a valid value there to recalculate.  What do you think? 

This happened when I removed the thickness there to put a new value. 

[image]

shaise, 2024-07-08:
You are definitely correct. Thanks for the catch.
Should be fixed now.

leoheck, 2024-07-08:
How do you manage these things on github? Do you just push to the master without changing the version? Or do you make branches? Package manager does not see updates. But updating manually I am seeing this.

[image]

leoheck, 2024-07-08:
This looks working on the current master, thanks. It is way better without red lines on the console.

closed 2024-07-08 by leoheck

shaise, 2024-07-08:
I update the package.xml as I was told. Dont know what the addon manager looks at

leoheck, 2024-07-08:
Yeah, it took time to update I guess, or maybe because I had changes on my copy of the repo.

shaise, 2024-07-08:
You are not the first one to complain, so I guess there is still an issue with that