shaise/FreeCAD_SheetMetal #355: Improvement - Base Shape Type settgins

leoheck, 2024-07-08:
Changing Base Shape type could hide the entire line instead of just disabling the input field. 
I think there is a property for that. 

What do you think? This would improve the usability a bit. 

[image]

leoheck, 2024-07-08:
[image]

Also, input fields with numbers could have more space...
The input field of the Base Feature could have the same min Width as the others some the fields could be the same length

shaise, 2024-07-08:
> Changing Base Shape type could hide the entire line instead of just disabling the input field. I think there is a property for that.
> 
> What do you think? This would improve the usability a bit.

I did not understand...

leoheck, 2024-07-08:
When I select `U-shape`, for instance. It disables the input field of `Flange width` as you can see the image above. The first image has the `Flat` as the Base shape which has the `Bend radius` and the `Height` disabled as well. 

Instead of disabling the input field keeping it there visible. The whole line could be hidden (removed) from the view.

shaise, 2024-07-08:
Now I understand. Hide instead of disable. Ok, I'll think about it.

zokhasan, 2025-07-31:
This issue did not close automatically, please close it

closed 2025-08-01 by luzpaz