shaise/FreeCAD_SheetMetal #356: Wrong update number for last update

lninio, 2024-07-12:
When updating the Sheet Metal workbench, the new version number matches the number of the last installed version.
[image]

lninio, 2024-07-12:
After installation, the newly installed version will be correct.
[image]

shaise, 2024-07-12:
Yes this is a commonly report issue. I need to check with the addon manager system if something was changed

luzpaz, 2024-09-10:
I haven't tested this but I'm not sure what is/was causing this issue. Everything looks in order:

https://github.com/shaise/FreeCAD_SheetMetal/blob/25c5f2b6df9af2fab63119d45c5e6bfe1cb28e4e/package.xml#L5-L6

> **\<date\>**
> REQUIRED
> The date of the current version, in the format YYYY-MM-DD or YYYY.MM.DD. 

https://wiki.freecad.org/Package_Metadata#.3Cdate.3E

shaise, 2024-09-10:
This is rather strange. Many times I get messages that the addon manager does not detect/updates the WB correctly, and the only solution is to uninstall/reinstall the WB.

luzpaz, 2024-09-10:
@shaise i could not find an open ticket for this issue on FreeCAD repo (https://github.com/FreeCAD/FreeCAD/labels/Mod%3A%20Addon%20Manager).

shaise, 2024-09-10:
The thing is it doesn't happen in other workbenches (or so I think) so I don't know if its an addon-manager thing.