shaise/FreeCAD_SheetMetal #357: Idea. The miters on the bends

lninio, 2024-07-12:
Implement automatic miters on bends if two bends with a relatively large radius are joined and there is not enough space at the junction for elongation. A typical example is the metal contacts of a 9V battery (6f22). In the image below, I have given the appearance of the contacts of this type of battery, as well as a possible view of the result of processing the coupling of bends in FreeCAD.

[image]
[image]

shaise, 2024-07-12:
Yes. Basically you want the existing Auto-Miter to apply also on the round part.

lninio, 2024-07-12:
Yes, it is. Otherwise, I get some redundant gaps around round areas. There are no necessary parameter values for the correct model.
[image]

lninio, 2024-07-12:
Another possible way is to use an "Extend face" and six separate bends for each "petal". But the result will still be wrong, because there is an intersection of bends.

[image]

pierreporte, 2024-07-15:
Would be useful for hems (see #302).