shaise/FreeCAD_SheetMetal #360: Fold Wall doesn't work if sketch is a clone/link

heeplr, 2024-08-08:
First of all let me thank you & all contributors for this **awesome** project!

I noticed that the "Fold a Wall" icon is disabled, when the sketch containing the fold line is a Link to a sketch (right click sketch in model tree -> Link actions -> Make Link") or the clone of a sketch (Draft WB -> Modification -> Clone).

Only "Make Wall", "Extend Face" and "Sketch On Sheet Metal" are available.

When having multiple bodies that should be folded on the same edge, it's convenient if you can just manage a single folding edge sheet instead of one copy for every body. If Links/Clones would work, all the other sheets would update automatically if the "master sheet" is changed.

shaise, 2024-08-08:
Thanks. I will take a look.

shaise, 2024-08-09:
@heeplr ,
can you zip and attach a test file?

heeplr, 2024-08-09:
Sure. Here's a project with two bodies (sheet001 and sheet002), one sketch with a mold line and the linked version of that sketch.

I generated the Fold of sheet001 and Sketch-fold like normal: select face, shift-select sketch, hit "Fold a Wall". Everything works.
But when I attempt the same thing with sheet002 and Sketch-fold002 (the linked version of Sketch-fold), the symptoms described above appear.

[fold-linked-sketch.zip](https://github.com/user-attachments/files/16558463/fold-linked-sketch.zip)

heeplr, 2024-08-09:
Just updated to the latest version and everything works like a charm now. Thank you!

heeplr, 2024-08-09:
whoops, premature celebration :(

While "Fold a wall" now works with linked sketches, it doesn't work with normal Sketch objects anymore.

It's reproducable by removing the Fold from my previous [example](https://github.com/user-attachments/files/16558463/fold-linked-sketch.zip) and trying to Fold it again. Link works, original Sketch doesn't.

shaise, 2024-08-09:
Ooops sorry about that...
Please uninstall and reinstall, because I left the same version.

closed 2024-08-09 by heeplr