shaise/FreeCAD_SheetMetal #361: Unfold fails on hinge shape

alexneufeld, 2024-08-12:
At least some of these test cases should be unfoldable, I think? They all fail.

[image]

```
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38318 (Git)
Build type: Unknown
Branch: main
Hash: a71f49f4f61a3849e69a99daf19c79f19431889d
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * Curves 0.6.39
  * OpticsWorkbench 1.0.18
  * sheetmetal 0.4.24
  * fasteners 0.5.24
  * WeldFeature 0.1.0
  * points_tools
  * OpenDark 2024.3.13
```
[sheetmetal_hinge_test.zip](https://github.com/user-attachments/files/16592283/sheetmetal_hinge_test.zip)

shaise, 2024-08-13:
The unfolder struggles when the edges are refined. I'm not familiar enough with the unfold code to fix it.
[image]

alexneufeld, 2024-08-25:
An update after doing some more testing:

I noticed that sometimes when mirroring this sort of shape and then refining it, faces get refined, but edges don't:

[image]

Cleaning up these edges sometimes helps, in addition to adding splitter edges along bends as suggested. All 4 of these test cases fail to unfold, though.

[image]
[hinge_test_2.zip](https://github.com/user-attachments/files/16741709/hinge_test_2.zip)

Here' a very crappy macro I've been working on with some functionality to re-split bend edges and/or clean up seam edges:

```python
import FreeCAD
import FreeCADGui
import Part
from itertools import combinations
from functools import lru_cache
from math import log10
from statistics import mode
from hashlib import md5

eps = FreeCAD.Base.Precision.approximation()

def round_vector(vec: FreeCAD.Vector, ndigits: int = None) -> FreeCAD.Vector:
    return FreeCAD.Vector(*[round(d, ndigits) for d in vec])

def estimate_thickness_from_edges(shp: Part.Shape) -> float:
    # assumes that the modal edge length equals the thickness of the sheetmetal part
    num_places = abs(int(log10(eps)))
    elist = [e.Length for e in shp.Edges]
    elist_rounded = [round(length, num_places) for length in elist]
    return mode(elist_rounded)

def estimate_thickness_from_deps(docobj: FreeCAD.DocumentObject) -> float:
    # find a basebend object that this depends on and use it's thickness
    return 1.0

def estimate_thickness_from_subshape(shp: Part.Shape, subshape: Part.Shape) -> float:
    # estimate thickness using a raycast across a selected face
    return 1.0

def estimate_thickness_from_cylinders(shp: Part.Shape) -> float:
    num_places = abs(int(log10(eps)))
    curv_map = {}
    for face in shp.Faces:
        if isinstance(face.Surface, Part.Cylinder):
            # normalize the axis and centerpoint
            normalized_axis = face.Surface.Axis.normalize()
            if normalized_axis.dot(FreeCAD.Vector(0, 0, -1)) < 0:
                normalized_axis = normalized_axis.negative()
            cleaned_axis = round_vector(normalized_axis, num_places)
            adjusted_center = face.Surface.Center.projectToPlane(
                FreeCAD.Vector(), normalized_axis
            )
            cleaned_center = round_vector(adjusted_center, num_places)
            key = (*cleaned_axis, *cleaned_center)
            if key in curv_map:
                curv_map[key].append(abs(face.Surface.Radius))
            else:
                curv_map[key] = [
                    face.Surface.Radius,
                ]
    # print(curv_map)
    combined_list_of_thicknesses = []
    for radset in curv_map.values():
        if len(radset) > 1:
            for r1, r2 in combinations(radset, 2):
                if (val := abs(r1 - r2)) > eps:
[... 120 lines cut ...]
                    # Part.show(surface_to_cut.toShape().generalFuse(wire)[0], "genfuse")
                    res = sorted(
                        surface_to_cut.toShape().generalFuse(wire)[0].Faces,
                        key=lambda f: abs(f.Area - face.Area),
                    )[0]
                    # Part.show(Part.makeCompound([surface_to_cut, wire]), f"CompoundToCut_{len(new_edges)}_{len(face.Edges)}")
                    # Part.show(res)
                    new_faces.append(res)
            else:
                new_faces.append(face)
        else:
            new_faces.append(face)

    return Part.makeSolid(Part.Shell(new_faces))

if __name__ == "__main__":
    sel = FreeCADGui.Selection.getSelection()[0]
    Part.show(unRemoveSplitter(sel.Shape))
    # Part.show(unRemoveSplitter(removeSplitterPoints(sel.Shape)))
    # Part.show(removeSplitterPoints(sel.Shape))
    # print(f"Estimated sheet metal thickness: {estimate_thickness_from_cylinders(sel.Shape)}")

```

I'll keep chipping away at this I think. Maybe I can come up with something to actually make the unfolder more robust.

PR #433 (2025-01-06): add new unfolder backend
This PR is to make sure we don't forget to add the new dependency in `package.xml` and close relevant issues. @shaise will plug in the backend functions to the GUI as needed.

This new unfolder:
- Uses the `networkx` python module for analyzing the sheet faces
- Should be indifferent to refined vs. unrefined shapes. Seam lines don't matter in most cases
- Is **5-50x faster than the old unfolder**  

closes #408
closes #361 
closes #345
closes #284 
closes #279 
closes #278 
closes #274 
closes #272 
closes #254 
closes #235 
closes #225 
closes #203 
closes #25 
closes [... cut]
closed 2025-01-06 by shaise