shaise/FreeCAD_SheetMetal #363: The hole disappears at the bend.

free777cat, 2024-08-20:
The hole disappears at the bend.

The hole disappears at the bend if the diameter is less than the length of the sweep and the hole does not extend beyond the edge of the bend.

[https://youtu.be/5vku80OS9XQ](url)

shaise, 2024-08-31:
Yes this is a tough one. not sure if its an easy fix.
@jaisekjames any ideas?

free777cat, 2024-09-04:
I experimented a little. 
SheetMetalBendSolid.py
def wrap_face(face, radius, axis, normal, zero_vert, center, zero_vert_normal):
The idea: to make a solid without bending (flat) using a similar method, subtract the solid of the middle part that already exists with a Boolean operation.
The result of the operation will be inverted holes (solid).

Bend two solids separately using the def wrap_face method.

Boolean operation: subtract the solid of the holes (bent from bent)

shaise, 2024-09-09:
@free777cat , thanks to @jaisekjames this should be fixed now. Please test and close the issue if works.
[image]

free777cat, 2024-09-09:
[image]

free777cat, 2024-09-10:
Now it is impossible to unfold
[image]
[ex2_.zip](https://github.com/user-attachments/files/16941522/ex2_.zip)

free777cat, 2024-09-10:
It may even be so...
[ex1_.zip](https://github.com/user-attachments/files/16941558/ex1_.zip)

shaise, 2024-09-10:
@free777cat , this is a different issue. Unfolding with holes in the bend did not work even before the current issue was fixed. I think this should be closed and a separate issue open for the unfolder. Mind, though, that this might take time since the unfold feature coder is not active.

closed 2024-09-10 by luzpaz

free777cat, 2025-01-06:
The latest version of SM has several bugs. This error returned again. Checked in v1.0, in v1.1dev, RT