shaise/FreeCAD_SheetMetal #371: Enhancement Request to add parametrized pipes

Ameritronik, 2024-09-15:
@ [Shai Seger](https://github.com/shaise),
Thank you for a great addition to Freecad.
Wondering if you could add a parametrized pipe, like you have the box, U section, angles etc.
Preferable the pipes would be defined as:
1. cross section of a regular polygon, or a circle
2. thickness of the sheet as separate entry
3. minimum gaps defined by sheet thickness, with user able to increase gap size
4. length of pipe would be center-axial line including any bends etc (probably harder to implement)
5. angles of pipes bent at distances from one end or arcs (if possible )
6. flanges at either end or one end.
7. minimum flanges width based on geometry computed by code, maximum set by user 
Probably a big list, but I think it would be useful for many HVAC duct works
Regards

shaise, 2024-09-15:
can you attach a sample of manually created such pipe? (say, a round one and a hexagonal one with flanges)

Ameritronik, 2024-09-16:
Hi,
I have enclosed a square cross section of an exhaust pipe which I did with
Sheet metal addon. Some edges were extended in sheet metal and then cut in
Part design to get the 45-degree angle.
I tried to extend this method to get a circular cross section, but it was too
Time consuming and so I gave up.
But if the enclosed (cannot unfold) can be included that would be a big help.
The Freecad version is 0.21.2 and the Sheetmetal version is 0.4.24
If you cannot see some parts of the object, use the view->visibility->show all objects.
Let me know if you can include this as an enhancement or if the addition is too time consuming etc, and so cannot be done. I am unable to help as I am
Not a coder.

Best regards
Hk

shaise, 2024-09-16:
I do not see the attachment. Perhaps you need to answer directly on github and not by mail.

luzpaz, 2024-09-16:
1. Click on this ticket: https://github.com/shaise/FreeCAD_SheetMetal/issues/371
1. zip up the example file
1. drag/drop in to github textfield

Ameritronik, 2024-09-16:
[OFD_v0.0.zip](https://github.com/user-attachments/files/17021014/OFD_v0.0.zip)

Ameritronik, 2024-09-16:
[image]