shaise/FreeCAD_SheetMetal #372: Bend perforation.   Errors

free777cat, 2024-09-20:
Perforation Angle  =   it is possible to enter a negative value

Perforation Initial Length  =   a null value cannot be entered !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

Is it still possible to add a perforation width parameter related to the Perforation Angle?
Changing one parameter recalculates the other and vice versa.
It is important when we use stamping with a tool of a certain size.

shaise, 2024-09-20:
@Erhannis , can you comment on that?

free777cat, 2024-09-20:
Perforation Angle =0 

maybe it should start from 90 degrees?

shaise, 2024-09-20:
> maybe it should start from 90 degrees?

I'm actually OK with that. 0 means the perforation adapts to the bend angle.

free777cat, 2024-09-20:
Then it would be a good idea to display the value of the angle. 
It will immediately be intuitively clear where it comes from.

shaise, 2024-09-20:
When it will have a special task panel for that it will be possible.

Erhannis, 2024-09-21:
1. Negative angle - I don't know whether or not that can be prevented with FreeCAD the way it is.  Somebody else with more FreeCAD programming experience would be better suited to answer that.  At the moment, it seems a negative angle just defaults to the same angle as the bend, which seems to me like reasonable behavior.  Taking the absolute value of the angle would also be a reasonable behavior, possibly more desirable.
2. Zero, you mean?  Oh, you're right.  Here's a fix: https://github.com/shaise/FreeCAD_SheetMetal/pull/374
3. Parameter dependencies - well, you could set both to formulas referencing a single cell in a spreadsheet, so they'd both update when you updated the spreadsheet.  I don't know whether or not FreeCAD directly supports linking two parameters bidirectionally; somebody else may know better.

free777cat, 2024-09-21:
Thank you! 

I am generally against the use of spreadsheets.

Erhannis, 2024-09-21:
> against spreadsheets

Oh?  I suppose that's allowed, haha, but I'm personally curious why that is.

free777cat, 2024-09-21:
In more complex projects, this is absolute horror!!!
If you correctly connect the dimensions of the parts on the sketches, you can always clearly and quickly find the original data.

Erhannis, 2024-09-21:
Huh, ok; good to know

closed 2024-09-22 by free777cat