shaise/FreeCAD_SheetMetal #377: make forming in wall    =   error

free777cat, 2024-10-04:
The file was created earlier

1. Loaded
2. After the update
[exmp.zip](https://github.com/user-attachments/files/17254303/exmp.zip)

[image]
[image]

luzpaz, 2024-10-11:
Please add your full [About](https://wiki.freecad.org/About) info

free777cat, 2024-10-12:
[code]
OS: Linux Mint 22 (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 2024.1006.0.14555 (Git shallow) AppImage
Build type: Release
Branch: (HEAD detached at 20241003stable)
Hash: 8058dcf52c2661edabf36d8147d5b8a187378d2d
Python 3.11.9, Qt 5.15.13, Coin 4.0.1, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Ukrainian/Ukraine (uk_UA) ]
Installed mods: 
  * FEMbyGEN 2.2.0
  * Sketcher
  * frame 0.1.1
  * Assembly4 0.50.6
  * mbdyn-dynamics
  * fasteners 0.4.72
  * lattice2 1.0.0
  * sheetmetal 0.4.26
[/code]

luzpaz, 2024-10-19:
@shaise can you reproduce ?

shaise, 2024-10-19:
@luzpaz , on my (long list). Just finished bringing Ondsel changes. So the list might grow even more...

shaise, 2024-10-20:
@free777cat it looks like you are using Linkstage freecad. I can not load your file under freecad main.

free777cat, 2024-10-20:
I was creating a new similar file = same result

free777cat, 2024-10-20:
https://github.com/realthunder/FreeCAD/releases

shaise, 2024-10-20:
> I was creating a new similar file = same result

please upload the file compatible with freecad main. Also if you know, please advise since which version it started to fail.

free777cat, 2024-10-21:
> Also if you know, please advise since which version it started to fail.

Last time it worked fine for me was 03/18/2024

free777cat, 2024-10-21:
[file.zip](https://github.com/user-attachments/files/17456690/file.zip)

Almost_impossible !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

OS: Linux Mint 22 (MATE/mate/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 (Git) AppImage
Build type: Release
Branch: main
Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * FEMbyGEN 2.2.0
  * QuickMeasure 2022.10.28
  * Sketcher
  * frame 0.1.1
  * Assembly4 0.50.6
  * mbdyn-dynamics
  * fasteners 0.4.72
  * lattice2 1.0.0
  * FEM_FrontISTR 0.1.0
  * DynamicData 2.56.0
  * sheetmetal 0.5.1
  * Defeaturing 1.2.2
  * dxf-library
  * Estimate 0.1.2
  * sheetmetal.backup1729443789.6656435 0.5.1 (Disabled)
  * ExplodedAssembly

free777cat, 2024-10-21:
The realthunder version does not work at all !!!

[code]
OS: Linux Mint 22 (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 2024.1006.0.14555 (Git shallow) AppImage
Build type: Release
Branch: (HEAD detached at 20241003stable)
Hash: 8058dcf52c2661edabf36d8147d5b8a187378d2d
Python 3.11.9, Qt 5.15.13, Coin 4.0.1, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 
  * FEMbyGEN 2.2.0
  * QuickMeasure 2022.10.28
  * Sketcher
  * frame 0.1.1
  * Assembly4 0.50.6
  * mbdyn-dynamics
  * fasteners 0.4.72
  * lattice2 1.0.0
  * FEM_FrontISTR 0.1.0
  * DynamicData 2.56.0
  * sheetmetal 0.5.1
  * Defeaturing 1.2.2
  * dxf-library
  * Estimate 0.1.2
  * sheetmetal.backup1729443789.6656435 0.5.1 (Disabled)
  * ExplodedAssembly
[/code]

10:36:13  Traceback (most recent call last):
  File "/home/Х/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalFormingCmd.py", line 153, in execute
    a = makeforming(tool, base, base_face, thk, tool_faces, offsetlist[i], fp.angle.Value)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/Х/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalFormingCmd.py", line 94, in makeforming
    offsetshell = tool.makeThickness(tool_faces, thk, 0.0001, False, False, 0, 0)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'Part.OCCError'>: 15StdFail_NotDone BRep_API: command not done
10:36:13  1.08e-07 <App> SheetMetalFormingCmd.py(488)|Document.cpp(3427): Failed to recompute Almost_impossible#WallForming: 15StdFail_NotDone BRep_API: command not done
10:36:13  Recompute failed!

shaise, 2024-10-21:
in `file.zip` it seems to work OK:
[image]

But Can you explain how you made the selections? Because the plate and the tool are in the same body, I can not see them together so I can not select both of them to create the forming.

free777cat, 2024-10-21:
[https://youtu.be/FIgpY6jECv8](url)

shaise, 2024-10-21:
Thanks!
Good to know. But I think this is very complicated way.
I try to separate it to 2 bodies, but I get an error when doing the forming. I dont know why, I need to check.

free777cat, 2024-10-22:
Do you have access to discord ondsel?
[image]

free777cat, 2024-10-22:
Ondsel has done a lot of interesting things lately.
But Ondsel is moving in its own direction, which is slightly different from the official FreeCAD branch.
The development of realthunder is more responsive to the needs of small and medium-sized businesses at this time.
Manufacturers are more interested in the stable operation of the software than the appearance of "super functionality" (which can break the operation at the most inopportune time).

https://ondsel.com/blog/gd-and-t-as-core-feature/

[image]

Academically, this is correct, but in ~90-95% of cases of work in the segment where FreeCAD is used, it will not be accepted in the production process.
More in demand is the simplification of modeling with the help of new advanced capabilities.
But the example of "make forming in wall" causes sadness.

shaise, 2024-10-22:
I Think realthunder's version of freecad is really the best. It has so many important features that just make everything much easier to work with. It is very disappointing that realthunder is not part of the main freecad.
The Ondsel version of sheetmetal was supposed to just add some Task windows to each operation to make it more usable. Unfortunately as I just discovered, it was not just the UI, they introduced a lot of bugs. I think in version 0.5.02 I finally removed all their bugs, and from now on, I will continue the UI myself.

Regarding this issue that we are trying to solve. 
I will download the latest realthunder version and see if indeed for older versions this function worked.

shaise, 2024-10-22:
Hi @free777cat ,
I downloaded the latest Realthunder freecad (like yours `Version: 2024.1006.0.14555 (Git shallow) AppImage`)
However when I try to load your file ([exmp.zip](https://github.com/user-attachments/files/17254303/exmp.zip)) I get
the following message:
[image]
When I click yes, I get the following error:
[image]

can you check and give me a file that loads without error?

free777cat, 2024-10-22:
The file was created in FreeCAD-Link-Stable-Linux-x86_64-py3.11-20240123.AppImage. In this version, it still opens without errors.

But I think the main thing is to make the "make forming in wall" tool work normally in the new version of the latest Realthunder freecad.
Let's just restore the normal operation of the tool.

free777cat, 2024-10-22:
Already in the next version (not the newest) of Realthunder freecad, I get errors when loading the file.

free777cat, 2024-10-22:
I don't think backwards compatibility is that important in this case.
No one complained except me. ;)

shaise, 2024-10-22:
Ok, I just tested.
In old realthunder freecad it works correctly. (see version below)
So new realthunder version it does not work. I think this is because of new OpenCascade library.
This error is also the same in new freecad main.
I will see if I can find a workaround for that.

```
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 2024.123.0.14555 (Git shallow)
Build type: Release
Branch: HEAD
Hash: 0c0f69ad4806682096b51f7669d5f1cb83f9c7d7
Python 3.8.18, Qt 5.15.6, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * BIM_Workbench 2021.12.0
  * fasteners 0.5.29
  * kicadStepUpMod 11.3.9
  * Manipulator 1.5.7 (Disabled)
  * OpenTheme 2024.9.1
  * PieMenu 1.8.0 (Disabled)
  * Render 2024.8.4
  * SelectorToolbar (Disabled)
  * sheetmetal 0.4.4
  * TabBar (Disabled)
[/code]
```

shaise, 2024-10-22:
There are some cased that freecad main also fails and display this error.
I have managed to make it work under main, but not under realthunder. I really don't know why. They both use the same OpenCascade version.

free777cat, 2024-10-23:
FreeCAD_weekly-builds-39024-conda-Linux-x86_64-py311.AppImage
Unfortunately, it doesn't work here

free777cat, 2024-10-23:
OS: Linux Mint 22 (MATE/mate/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git) AppImage
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

shaise, 2024-10-23:
Sorry, can you attach the file that is not working on main? Because I tested the old ones and it worked.
(For some reason the form tool is problematic. I use 2 different codes, if one fails I use the other.)

shaise, 2024-10-23:
Here is another strange observation done on realthunder version:
If I try to make a from where the tool is in another body, it fails.
But if I make a copy of the tool, and put it outside the body, it works good.
see attached:
[check_forming_issue.zip](https://github.com/user-attachments/files/17491133/check_forming_issue.zip)

shaise, 2024-10-23:
Ok, I have went down to the main difference between main and realthunder. A simple "Part->Thickness" is working on main and not realthunder. I will post an issue for him.
shai.

[exmp_thickness.zip](https://github.com/user-attachments/files/17497341/exmp_thickness.zip)

shaise, 2024-10-23:
By changing the tool a little bit, it works with your example:
[exmp_fixed.zip](https://github.com/user-attachments/files/17491557/exmp_fixed.zip)

shaise, 2024-10-23:
I have submitted an issue: https://github.com/realthunder/FreeCAD/issues/1039

free777cat, 2024-10-24:
Thank you!  At this time, I see progress.

free777cat, 2024-10-24:
I want to suggest a little change in the logic of work.
From work experience, it would be more clear and would certainly help to avoid mistakes in further development.

1. We create two bodies: the working "body" and the tool
2. In the working "body", we create a sketch ("placement sketch")
3. In the working "body", we create a subshape binder of the tool
4. Based on the "placement sketch" (first click), add the planes of the "subshape binder" tool (second, third, .... clicks)
5. Perform "Make Forming"

Everything will be in one body and everything will be visible. Placement of the required amount of "Make Forming" is also carried out immediately.