shaise/FreeCAD_SheetMetal #378: Make bend : wrong radius on opposite edge

rmdug, 2024-10-04:
Hello,

I’m new to FreeCAD and discovering the sheetmetal workbench.
If I try to generate a bend from an offsetted 4 mm thick angle : 
Initial shape : 
[image]

Result after setting bend radius to 3 mm on the inner edge : 
[image]

The issues are : 
+ The radius on the external edge should be 7 mm but is the same as the inner radius (3 mm) ;
+ A parasite surface appears at both ends of the bend.

Here is my config : 

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/Belgium (fr_BE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Assembly3 0.12.2
  * Behave-Dark-Colors 0.1.1
  * Beltrami 1.1.0
  * Curves 0.6.48
  * Defeaturing 1.2.2
  * Dracula 0.0.5
  * ExtremeProDark 2.9.0
  * freecad.gears 1.3.0
  * ModernUI
  * pcb 6.2023.1
  * ProDarkThemePreferencePack 1.0.0
  * sheetmetal 0.4.26
  * Silk 0.1.5

Thank you for your great tool !

Rémi

shaise, 2024-10-04:
1. make sure you select the outer edge when performing this command
2. the extra edge on the side can confuse the tool:
[image]

rmdug, 2024-10-04:
Thank you for your fast answer !

1. Unfortunately the bend won’t work from the outer edge : 
[image]
[image]

After applying the bend the whole body disappeared and I had to activate and deactivate the feature suppression to bring the body back, with the displayed error.

2. I think this extra edge was generated by the offset function in the part workbench.

[2024-09-25-Bend from offsetted body.zip](https://github.com/user-attachments/files/17258309/2024-09-25-Bend.from.offsetted.body.zip)

shaise, 2024-10-04:
it seems that the offset tool did not create a valid solid, hence the sheetmetal tool could not handle it. The best way to convert a line sketch to a sheetmetal object is using the BaseBend tool directly on the sketch:
[image]
This uses much less steps then your approach.

see object 2 in the attached file
[2024-09-25-01-modified.zip](https://github.com/user-attachments/files/17259044/2024-09-25-01-modified.zip)

rmdug, 2024-10-07:
Thank you for your answer. Indeed your way is much more efficient from a sketch : )

With the commercial CAD software I use daily (SolidWorks, Inventor, Catia) I often prepare surface body for tricky sheetmetals before turning them to solid/sheetmetal, so that I don’t have to deal with the artifacts due to the material thickness.

So my intention with this test was to see how the sheetmetal workbench handles surfaces. As the WB doesn’t allow the generation of a sheetmetal from a surface body I tried from an offsetted body, which seemed to be handled by the workbench.

Should I rename this thread, or even change it from « issue » to « feature request » ?

shaise, 2024-10-09:
The sheetmetal can not handle these type of objects and will not in the future since the object was not a legal solid. 
You can definitely set a feature request for sheetmetal to turn a surface directly from a surface (without the need for offsetting it first)

rmdug, 2024-10-10:
OK, I just created this feature request : https://github.com/shaise/FreeCAD_SheetMetal/discussions/380

and close this non-issue. Thank you for your help !!

Rémi

closed 2024-10-10 by rmdug