shaise/FreeCAD_SheetMetal #379: Issue in Sheet Metal Workbench 0.4.26

salvatore28, 2024-10-04:
clicking on FoldWall adds the Fold in the tree view but doesn't work and produces errors:

[image]

Reverting to version 0.4.24 resolves the problem.

OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Italian/Italy (it_IT)
Installed mods: 
  * kicadStepUpMod 11.3.9
  * QuickMeasure 2022.10.28
  * sheetmetal 0.4.26

salvatore28, 2024-10-04:
Some other informations that could be useful:

In another PC I still have version 0.4.24 of SheetMetalWorkbench that I have not updated and, for this reason, all works as expected and without errors.

Someone suggested, to revert to an older version, to simply overwrite the files in "%appdata%\FreeCAD\Mod\sheetmetal" with the old files of version 0.4.24. This doesn't work.

Doing so in this PC and stopping updates makes the addon manager to believe that 0.4.24 is installed, infact the old icon is shown; but trying to add the fold in the project after having selected the face and the bend line, results in almost all icons of sheet metal workbench to be grayed and so unusable. Probably, wiping the sheetmetal folder and manually substitute it with the old version folder is not appropriate maybe because the addon is not properly installed in this way.

shaise, 2024-10-04:
can you attach the file?

salvatore28, 2024-10-07:
> can you zip and attach the file?

hello, sorry for the delay. Here is the file of the project, but I suppose that any basic project with a foldwall could be adequate:

[Alimentatore - Pannello anteriore.zip](https://github.com/user-attachments/files/17274836/Alimentatore.-.Pannello.anteriore.zip)

Regarding the impossibility to correctly downgrade addons, I think that this could be a great enhancement for the addon manager; in addition, another great enhancement could be the possibility to install any addon offline, that's to say from a local zip file. Maybe you could urge for this with FreeCAD's authors.

Luckily, I have a third FreeCAD installation on my notebook with an even older version of SheetMetal Workbench (0.4.11) that allowed me to finish the file for the workshop.

Regards
Salvatore

salvatore28, 2024-10-14:
hello,

waiting for the resolution of this problem, can you suggest a secure method to downgrade to a previous version kept into a zipped file?

Thank you
Salvatore

luzpaz, 2024-10-14:
>  Version: 0.21.2.33771 (Git)

@salvatore28 can you test on RC2 since FreeCAD is about to release version 1.0 ?  
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc2

salvatore28, 2024-10-14:
> > Version: 0.21.2.33771 (Git)
> 
> @salvatore28 can you test on RC2 since FreeCAD is about to release version 1.0 ? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc2

of course, but is RC2 installed in a different folder or does it overwrite existing installation?

Salvatore

salvatore28, 2024-10-14:
whops, there is no installer at all, I must overwrite existing files. Ok, I do it. Reinstalling the latest stable version is quick.

Salvatore

salvatore28, 2024-10-14:
good news. Using RC2 SheetMetal Workbench 0.4.26 works as expected. Finally, it was a problem in FreeCAD and not in the extension, right?

Thank you.

Salvatore

salvatore28, 2024-10-21:
> I can confirm it works on RC2 and SheetMetal 5.02

yes. I have updated to RC2 all the three installations of FreeCAD and they all work, still with 0.4.26.

Is the release date of FreeCAD 1.0 alredy known?

Thank you.
Salvatore

salvatore28, 2024-10-21:
do I have to close this issue?

Sal

shaise, 2024-10-21:
Please close it. You can always re-open it if the problem repeats.

closed 2024-10-21 by luzpaz