shaise/FreeCAD_SheetMetal #381: BaseShape doesn't fill gaps

rmu75, 2024-10-11:
It seems BaseShape is missing some properties from SmBendWall like "Auto Miter Minimum Gap" or "Auto Miter maximum Extend Distance" which leads to somewhat puzzling behaviour. Is there some rationale or would a patch be welcome that fixes that?

luzpaz, 2024-10-11:
Please add your full [About](https://wiki.freecad.org) info so we know what version of FreeCAD and  SheetMetal you're using

rmu75, 2024-10-11:
OS: KDE neon 6.1 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38895 (Git)
Build type: Unknown
Branch: releases/FreeCAD-1-0
Hash: 6c64197cdb9ec85a81b854f6f06c33e78e49d936
Python 3.10.12, Qt 5.15.14, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Austria (de_AT)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * btl 0.9.9
  * kicadStepUpMod 11.3.9
  * dodo 1.0.1
  * QuickMeasure 2022.10.28
  * OpticsWorkbench 1.0.24
  * OpenTheme 2024.9.1
  * Assembly3 0.12.2
  * frame 0.1.1
  * sheetmetal 0.5.2
  * FeedsAndSpeeds 0.5.0
  * offline-documentation 1.0.0-alpha
  * InventorLoader 1.4.0
  * EM 2.1.1

rmu75, 2024-10-21:
Example demonstrating the issue. Base shape, tub, 2mm thickness, 3mm radius vs. 3.01mm radius.

[example.zip](https://github.com/user-attachments/files/17465367/example.zip)

OS: KDE neon 6.2 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Austria (de_AT)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * sheetmetal 0.5.2
  * WB_Organizer 2024.1.29
  * btl 0.9.9
  * CamScripts

rmu75, 2024-10-21:
[image]

shaise, 2024-10-21:
Thanks, I will investigate. This is not an auto miter issue. auto miter is only on hat and box.

rmu75, 2024-10-21:
it's the extend distance property default of 5mm that can't be changed that leads to that. 

https://github.com/shaise/FreeCAD_SheetMetal/blob/master/SheetMetalCmd.py#L884

I think the Base object needs a property like
https://github.com/shaise/FreeCAD_SheetMetal/blob/master/SheetMetalCmd.py#L1489
and probably related stuff.

shaise, 2024-10-22:
I see. Thanks. Then for the base object, I will just auto update this value to always work, since either you want it or not. I don't want to over complicate the base shape.

rmu75, 2024-10-22:
Yes that makes sense. I can work on that if you want.

shaise, 2024-10-22:
With the latest version it should work now.

luzpaz, 2024-10-25:
@rmu75 can you reproduce with the fix ?

rmu75, 2024-10-28:
Thanks -- seems to work.