shaise/FreeCAD_SheetMetal #385: Create sheetmetal from surface body

luzpaz, 2024-10-20:
### Discussed in https://github.com/shaise/FreeCAD_SheetMetal/discussions/380

<div type='discussions-op-text'>

<sup>Originally posted by **rmdug** October 10, 2024</sup>
This proposal is linked to this non-issue : https://github.com/shaise/FreeCAD_SheetMetal/issues/378

In sheetmetal design it can be very handy to create first a surface body and then turn it to a sheetmetal:
+ surface creation
For example:
[image]

+ sheetmetal creation:
    + set thickness;
    + set bend data (radius, K factor,…);
    + set corner data and free edges joints,

+ selection of bent edges
For example (bent edges in red):
[image]

This way of doing proved to be very efficient to do tricky shapes including unusual bend angles with the commercial software I use.
Thank you for your great work : )</div>

EstebanFPV, 2024-11-27:
+ 1 I would greatly appreciate this being a feature. square to round transitions even square to square or round to round transitions are nigh impossible to make with the sheetmetal workbench, but with Alibre's built in one, they are a breeze.

ma00aem, 2025-01-13:
We use this type of workflow in another cad system (multibody sheet metal). It is fantastic as you can have one reference part and everything follows that part.

shaise, 2025-01-13:
On my list.

Mannshoch, 2025-03-10:
In Solidworks there are two approaches for that. The [old one](https://help.solidworks.com/2025/english/SolidWorks/sldworks/t_add_body_Insert_bend.htm), which you usually use to convert an imported sheet-metal (e.g. STEP, Or you have to create gaps manually before) 

And the [modern one](https://help.solidworks.com/2025/english/SolidWorks/sldworks/t_add_body_Convert_to_SM.htm) which allow to select a surface and let you [select](https://help.solidworks.com/2025/english/SolidWorks/sldworks/HIDD_DVE_CONVERT_TO_SM.htm?id=9ffce838f4a042af8d67c6ecba76e1fd) the bows or edges to the next wall. (This tool detect needed gaps automatically and give the possibility to design the corner relief on this detected gaps). This Modern solution allow also to take several Sheets from the same Body.

Mannshoch, 2025-03-10:
Long ago, on a CAD without sheet-metal capabilities, I used a "bent less" sheetmetal design. There I focused on designing a Sheetmetal with corners or I worked with a virtual edge (where the walls would meet) and designed corner relief (Surface body for cut) and Junction and so one. Everything I created was designed from this edge and at the end I rounded everything with fillet. Big plus, that way I had all bent radius at the end of the feature tree and where able to change at any time and had a parametric design that let me change values fast. 

In a way, here you would do it the same way but by Macro.

shaise, 2026-03-14:
technically no, since the feature is solid to sheet and not surface to sheet. I will investigate how hard to support both.