shaise/FreeCAD_SheetMetal #386: I'm sorry, of course, but some bullshit happened (so says one little person)

free777cat, 2024-10-20:
"Add base shape" creates two bodies
realthunder/FreeCAD/releases  Stable 2024.10.03

[image]
[image]

shaise, 2024-10-20:
He He! nice redundancy.... :)
Ok I'm on it.

shaise, 2024-10-20:
Fixed. It seems Ondsel's version only permits base shape inside a body.

free777cat, 2024-10-20:
Fold a Wall
1.Improper location of the fold line
2.Sketch doesn't hide
[image]
[image]
[image]

free777cat, 2024-10-20:
Ondsel's version
I would be very careful using this!!!

[https://discord.com/channels/1180269060169154620/1197656610361843812/1284566314899341436](url)

[image]

shaise, 2024-10-20:
This is very disturbing. I wish I knew it before.

luzpaz, 2024-10-20:
@free777cat do you mind posting the full About info of the versions you're reporting about? It just makes triage easier. TIA

free777cat, 2024-10-20:
https://github.com/realthunder/FreeCAD/releases

[code]
OS: Linux Mint 22 (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 2024.1006.0.14555 (Git shallow) AppImage
Build type: Release
Branch: (HEAD detached at 20241003stable)
Hash: 8058dcf52c2661edabf36d8147d5b8a187378d2d
Python 3.11.9, Qt 5.15.13, Coin 4.0.1, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 
  * FEMbyGEN 2.2.0
  * QuickMeasure 2022.10.28
  * Sketcher
  * frame 0.1.1
  * Assembly4 0.50.6
  * mbdyn-dynamics
  * fasteners 0.4.72
  * lattice2 1.0.0
  * FEM_FrontISTR 0.1.0
  * DynamicData 2.56.0
  * sheetmetal 0.5.1
  * Defeaturing 1.2.2
  * dxf-library
  * Estimate 0.1.2
  * ExplodedAssembly
[/code]

free777cat, 2024-10-20:
> This is very disturbing. I wish I knew it before.

This is the file that the user writes about (this is not my post)
[file.zip](https://github.com/user-attachments/files/17450109/file.zip)

shaise, 2024-10-21:
Thank you. This bug was fixed. Please uninstall and reinstall sheetmetal. I did not change the version yet.

free777cat, 2024-10-21:
When creating BaseBend 
[image]
Sketch doesn't hide

shaise, 2024-10-21:
Version 5.0.2 should fix the baseband not hiding.

free777cat, 2024-10-22:
If the topic is about all kinds of bullshit...
Should a picture be added to the description?

https://support.ptc.com/help/creo/ced_modeling/r20.5.0.0/en/index.html#page/ced_modeling/OSDM_Main/SheetParts_KFactor.html

[image]

free777cat, 2024-10-31:
Have you already seen the news about the closure of Ondsel?

shaise, 2024-10-31:
I have heard very little. Are they closing the entire project? Is there an announcement anywhere?

free777cat, 2024-10-31:
https://ondsel.com/blog/goodbye/

free777cat, 2024-10-31:
They had plans to create "built-in" SheetMetal - an idea that I personally did not like very much...

shaise, 2024-10-31:
They wanted both sheetmetal and fasteners built in. These are the most used external workbenches. (much more then some internal ones - like robot)

free777cat, 2024-10-31:
Couldn't this lead to a certain "monopoly" of sheet metal and fasteners from the point of view of code development?
In my opinion, this would reduce the openness of the code and the speed of responding to problems.

shaise, 2024-10-31:
I'm not concerned about the "monopoly". The non-monopoly of assembly WB was in my opinion a problem with so many versions.
However I AM concerned about the development and the responsiveness to issues.
That why I think it should be like blender add-ons. The popular ones should be bundled with the release, but you can always disable/enable/update it via the addon manager.

free777cat, 2024-10-31:
> The popular ones should be bundled with the release, but you can always disable/enable/update it via the addon manager.
That would be very correct. There are people who have no idea about sheetmetal.

free777cat, 2024-11-22:
Look at this

shaise, 2024-11-22:
Thanks. Working on it

shaise, 2024-11-22:
Sorry about that. still issues from the merging with Ondsel.

luzpaz, 2024-12-04:
@free777cat can you rename this ticket to be more relevant and descriptive of the issue ?

closed 2024-12-05 by free777cat