shaise/FreeCAD_SheetMetal #387: Unfold_Sketch Problem

lryng, 2024-10-20:
When performing the Unfold function with the Sketch option. Sketch is rotated 180 in the wrong axis.
[image]
[Sheet Metal Unfold_Sketch Problem.zip](https://github.com/user-attachments/files/17451506/Sheet.Metal.Unfold_Sketch.Problem.zip)

shaise, 2024-10-20:
please add freecad info (help -> about, then press "copy to clipboard")

lryng, 2024-10-21:
[image]

shaise, 2024-10-21:
Strange, I use the same freecad version but I can not load the example file (as if its from old version, or realthunder linkstage version)

lryng, 2024-10-22:
I tried again to make it unfold and right now everything is okay.
Sorry for the problem 
[image]

closed 2024-10-25 by luzpaz