shaise/FreeCAD_SheetMetal #389: Sheetmetal operations do not update on complex model

schauma, 2024-10-27:
I had a fairly complex model when I realized my dimensions needed to be changed.
To be specific, I have a thin Box that I used as a "base" for various bends/pads/pockets that I want to change the dimensions of.
I have all my dimensions in a spreadsheet parametrized. However when I change the values, the model does not update. Even if I manually entered the values without referencing the Spreadsheet, the model does not change.
I thought I tested this beforehand and indeed, if I do the same changes on a much more simplified model it works as expected.

So now, why are the changes in the simple model applied while the more complex model does not change? Is there a methodology that is followed?
Are there certain operations/features that break the recomputation process with sheetmetal WB?
I tried to recompute every step individually however there is no change whilst also no error. 

If needed, I can supply my freeCAD file.

I tested this on the latest stable FreeCAD version, the 1.0rc2 and the weekly build, without success.
Sheetmetal workbench 0.5.0

luzpaz, 2024-10-27:
Latest sheetmetal version is 
```
V0.5.03 22 Oct 2024: Fix fill gap issue in base shape feature. - Partially Fix forming issue not working in latest freecad
```

Can you always provide your full [About](https://wiki.freecad.org/About) info when you open a ticket?  
Also can you provide an example file? Zip up you .fcstd file and drag/drop it into Github textbox to it auto-attaches to this ticket. TIA

schauma, 2024-10-27:
```
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) Snap 908
Build type: Release
Branch: tag: 0.21.2
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.7, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Curves 0.6.51
  * sheetmetal 0.5.3
  * A2plus 0.4.68

```

Thanks for the fast reply. I do not know how to specifically export that part in so I just compressed the whole thing. My problem specifically refers to my "Frame" Body and directly the dimensions of the very first box.
I hope this helps

[old_pc.zip](https://github.com/user-attachments/files/17533729/old_pc.zip)

shaise, 2024-10-27:
I can confirm this issue, but I dont know if it is sheetmetal related. When I change values, the sheetmetal execute function is not being called at all.
Also @schauma , I'm curious, the base box is is 310x1x180 where the width is 1 not the height. So how come in the view the width is on the z axis and not the y axis? i did not see any attachments/rotations:
[image]
when I try to create a new box in a new body with exact same dimensions, they plate is standing, not laying flat:
[image]

schauma, 2024-10-27:
I cannot give you a definitive answer. This file has been opened and modified in 3 different FreeCAD versions so my suspicion would be that maybe something got mixed up in there. I also remember struggling with rotations of the different coordinate systems at one point. But as I am still a beginner I cannot pin-point where that manifests itself.
I double-checked in an empty file as well and I get the same behavior you showed. I really hope this is not the cause for this issue. I'd hate to redo the same thing once more. 
Do you see any bugfix at the horizon for this problem or can you recommend a workaround by any change?

shaise, 2024-10-31:
I tried many things and couldn't figure out why the parts are not being updated. Perhaps someone in the freecad forum have encountered this issue so maybe try to post the problem there as well.

sheetmetalman, 2024-11-17:
I think maybe anyone open this file in a FreeCAD without Sheet Metal workbench and saved the file. This make the object lose the parameterization.

No class is assigned to the objects. As shown in the image below.
Instead of "None" it should say something like "SheetMetalCmd.SMBendWall".

[image]

I try here assign the correct class to the first Bend object and it works. Maybe this could be an workaround.
Here is the code I used to do this. It's just a example where I assign the class again to the "Bend" object.
(Erase the object properties is needed, because when try to assign the class to object again, isn't allowed to overwrite the properties. So, to use this workaround, you have to apply the properties to the objects again.)

```
# Get the object
obj = App.ActiveDocument.Bend
selobj = App.ActiveDocument.Box

# Erase the properties
for propName in obj.PropertiesList:
	obj.removeProperty(propName)

# Assign the class to the object again
import SheetMetalCmd
obj.Proxy = SheetMetalCmd.SMBendWall(obj, selobj, "Edge7")

```

shaise, 2024-11-17:
@sheetmetalman , thanks for the catch!
But this mean you had to do it one by one to all sheetmetal's features?

sheetmetalman, 2024-11-17:
Yes.

shaise, 2024-11-17:
If what you say is true, and links disappear when a WB is missing, then freecad must warn if a file is saved without the appropriate WB.

sheetmetalman, 2024-11-17:
I didn't test if this happened. It was just a guess.

sheetmetalman, 2024-11-17:
But I test it now. And it's true.

In the zip file have a file saved with SheetMetalWB and another file opened in a FreeCAD without SheetMetalWB and saved.

[test losing class.zip](https://github.com/user-attachments/files/17791465/test.losing.class.zip)

I got the same result: the object is not assigned to a class.

[image]

schauma, 2024-11-17:
@sheetmetalman Thank you for all your insightful research. I have to admit that most of what you are describing is way above my head, but I am very thankful that you tried to fix my project. I will have to check it in the next couple of days.  
Maybe as I side question, how did you get to know all of this, are there any resources on how to do proper FreeCAD debugging?

sheetmetalman, 2024-11-17:
@schauma. You're welcome! I'm glad I could help.

I discovered this because I saw a similar problem recently.

A very helpful tool is asking to ChatGPT. Know the FreeCAD code logic is very helpful for debugging. My tip is learn python programming for FreeCAD, do some exercises, learn about programming (what help me was learn what is "OOP" and "class").