shaise/FreeCAD_SheetMetal #390: Add issue template to bugtracker

luzpaz, 2024-10-27:
https://docs.github.com/en/communities/using-templates-to-encourage-useful-issues-and-pull-requests/configuring-issue-templates-for-your-repository

shaise, 2024-10-27:
Nice!
Is there a ready template I can use?

luzpaz, 2024-11-09:
Modify the existing upstream template? https://github.com/FreeCAD/FreeCAD/blob/main/.github/ISSUE_TEMPLATE/PROBLEM_REPORT.yml

shaise, 2024-11-09:
Thanks! I will take a look

hasecilu, 2024-11-11:
Hi, recently I also created them for my WBs. Here is a example: https://github.com/bilbaomakers/StemfieWB/commit/f65f169208173624994134a15ddc5227ba81fa11

shaise, 2024-11-17:
@luzpaz , @hasecilu 
Thanks for your help! 
I have added the bug report / feature request templates.

luzpaz, 2024-11-27:
Well Done!

closed 2024-11-27 by luzpaz