shaise/FreeCAD_SheetMetal #391: de/selection of flange parameters not working

JaapStruyk, 2024-11-07:
not possibile to change or delete selected edges for a bend in Flange Parameters, not by clicken and neither by the delete keystroke.

OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * fcgear
  * fasteners
  * dxf-library
  * CfdOF 1.27.14
  * sheetmetal 0.5.3
  * Curves 0.3.0
  * Manipulator 1.5.7

shaise, 2024-11-07:
I do not understand the problem. can you attach an image and also a sample file?

JaapStruyk, 2024-11-07:
sample file not needed anything with multiple bends will do, not possible to deselect edges neither with mouse or keyboard, adding does work.
[image]

shaise, 2024-11-07:
Ok I understand.
There seems to be a small display bug there, but it is still functional.
Press the (select) button.
You should see the original edges that was there. if you click on a new edge, it will be added to the list, if click on a selected edge it will be removed from the list. you then need to click (preview) to show the new list and update the model. Wile selecting/deselecting the view is not updated correctly, this indeed is a bug and I will try to solve it, however it will still work even if not shown.

shaise, 2024-11-07:
It seems to be a general freecad bug not related to sheetmetal
I have posted an issue: https://github.com/FreeCAD/FreeCAD/issues/17732

shaise, 2024-11-08:
@JaapStruyk ,

This issue was apparently fixed 2 weeks ago. So I guess it will be available in the next release candidate.
For now, you can download the latest [weekly build ](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)where this issue is already fixed.

luzpaz, 2024-11-08:
@JaapStruyk there are so new features in the weeklies, so no.

luzpaz, 2024-11-08:
Closing ticket as already fixed upstream

closed 2024-11-08 by luzpaz