shaise/FreeCAD_SheetMetal #395: Extruded Cutout does not work in realthunder version

free777cat, 2024-11-13:
Extruded Cutout does not work in realthunder version

09:56:33  Running the Python command 'SheetMetal_AddCutout' failed:
Traceback (most recent call last):
  File "/home/Хата/.local/share/FreeCAD/Mod/sheetmetal/./ExtrudedCutout.py", line 382, in Activated
    SMBody = selected_object.getParent()
             ^^^^^^^^^^^^^^^^^^^^^^^^^

'FeaturePython' object has no attribute 'getParent'

[code]
OS: Linux Mint 22 (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 2024.1006.0.14555 (Git shallow) AppImage
Build type: Release
Branch: (HEAD detached at 20241003stable)
Hash: 8058dcf52c2661edabf36d8147d5b8a187378d2d
Python 3.11.9, Qt 5.15.13, Coin 4.0.1, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 

  * sheetmetal 0.5.4

[/code]

shaise, 2024-11-13:
Is it same as #392?

free777cat, 2024-11-13:
I couldn't find #392 today. Decided that this post was not published.
Added error view

shaise, 2024-11-13:
Try now

free777cat, 2024-11-13:
Everything is fine now. Thank you!

closed 2024-11-13 by shaise

free777cat, 2024-11-13:
In my opinion, the function icon is very uninformative.
Maybe the look like in the picture better reflected the idea of ​​the function.
[image]

shaise, 2024-11-13:
It difficult to make that small

free777cat, 2024-11-13:
I tried in GIMP.
If you lower the existing one a little, there should be enough space.   Visually it looks clearer.

free777cat, 2024-11-13:
Extruded cutout (normal to thickness) from extrusion sketch.

Such an explanation will more accurately reflect the essence of the function.