shaise/FreeCAD_SheetMetal #398: Unfold  with material_foo problem

MuratUelker, 2024-11-16:
[Unnamed.zip](https://github.com/user-attachments/files/17784354/Unnamed.zip)

After last two updates, i get this error if i try to unfold sheetmetal object with material_foo spreadsheet. I get this error from linux, mac and windows.

13:38:48  pyException: Traceback (most recent call last):
  File "/home/red/.local/share/FreeCAD/Mod/sheetmetal/./UnfoldGUI.py", line 210, in accept
    params = self._getData()
             ^^^^^^^^^^^^^^^
  File "/home/red/.local/share/FreeCAD/Mod/sheetmetal/./UnfoldGUI.py", line 142, in _getData
    lookupTable = SheetMetalKfactor.KFactorLookupTable(
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/red/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalKfactor.py", line 94, in __init__
    key_cell = self.find_cell_by_label(lookup_sheet, "Radius / Thickness")
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/red/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalKfactor.py", line 132, in find_cell_by_label
    for cell in self.get_cells(sheet):
                ^^^^^^^^^^^^^^^^^^^^^
  File "/home/red/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalKfactor.py", line 123, in get_cells
    return sorted(filter(self.cell_regex.search, sheet.PropertiesList))
                                                 ^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'str' object has no attribute 'PropertiesList'

luzpaz, 2024-11-16:
Missing full [About](https://wiki.freecad.org/About) info

MuratUelker, 2024-11-17:
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0RC4.39104 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc4)
Hash: 83646753e051eaf438c0d5e9351cb57bd27377ab
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * sheetmetal 0.5.5
  * OpenTheme 2024.9.1
  * PieMenu 1.9.5
  * QuickMeasure 2022.10.28

shaise, 2024-11-17:
Thanks. This is fixed now.

MuratUelker, 2024-11-18:
Problem solved. Thanks for the help

shaise, 2024-11-19:
great! closing.

closed 2024-11-19 by shaise