shaise/FreeCAD_SheetMetal #399: Unfold Extruded Cutout

free777cat, 2024-11-18:
### Describe your issue

Unfold Extruded Cutout

11:57:39  exception at line 3128("cannot access local variable 'nodeFace' where it is not associated with a value",)
11:57:39  ("cannot access local variable 'nodeFace' where it is not associated with a value",)

### FreeCAD version info + SheetMetal WB version

```shell
Hash: 8058dcf52c2661edabf36d8147d5b8a187378d2d
Python 3.11.9, Qt 5.15.13, Coin 4.0.1, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 
 

  * sheetmetal 0.5.6

[/code]
```

### Put here any screenshots or videos (optional)

[unfold.zip](https://github.com/user-attachments/files/17798535/unfold.zip)

sheetmetalman, 2024-11-18:
Hi!

It's because the `.removeSplitter()` function.

I modified the code and temporaly add an option to remove the `.removeSplitter()` and, after this, the unfold works.

@free777cat, as a workaround, if you want to do this unfold as soon as possible and feel safe to modify the code, you can remove all 3 `.removeSplitter()` in the ExtrudedCutout.py in your computer, update the ExtrudedCutout object in you model, and the unfold gonna work.

@shaise should I add and option (like the image below) to remove the `.removeSplitter()` or remove it forever from the code?

[image]

shaise, 2024-11-18:
As in some other functions, it is best to have a Refine function.
please note that if you add a new property, if you load an old file this property will not be shown and might cause errors. 
To solve this you need to change your code to use the `SheetMetalTools.smAdd<X>Property` system.
see how it works on SheetMetalCmd.py file. This checks if the propery exists, If not it will add it
note that self._addProperties(obj) is used in 2 places

closed 2024-11-21 by free777cat