shaise/FreeCAD_SheetMetal #401: Extruded Cutout    Incorrect work

free777cat, 2024-11-20:
### Describe your issue

     Extruded Cutout    Incorrect work

### FreeCAD version info + SheetMetal WB version

```shell
[code]
OS: Linux Mint 22 (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 2024.1006.0.14555 (Git shallow) AppImage
Build type: Release
Branch: (HEAD detached at 20241003stable)
Hash: 8058dcf52c2661edabf36d8147d5b8a187378d2d
Python 3.11.9, Qt 5.15.13, Coin 4.0.1, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 

  * sheetmetal 0.5.6

[/code]
```

### Put here any screenshots or videos (optional)

[Incorrect_work.zip](https://github.com/user-attachments/files/17827539/Incorrect_work.zip)
[image]

shaise, 2024-11-20:
@sheetmetalman , can you take a look?

sheetmetalman, 2024-11-20:
Yes! As soon as possible.

sheetmetalman, 2024-11-22:
Hi!

As I understood, the problem is that the result geometry is entering the sketch extrusion zone, as show in the image below.
But the correct geometry should be the second image.

[image]
[image]

Looking other software (SolidWorks) at how they deal with, I find they have the same issue, BUT they implement a function to resolve this. Solidworks calls it "Optimize geometry". The below images and video show what happen in this case in SolidWorks.

[image]
[image]
[image]
[image]

From where I take the pictures above: https://www.youtube.com/watch?v=_VdK-DPmJQo and https://www.javelin-tech.com/blog/2017/01/solidworks-2017-normal-cut-optimize-geometry/

I implement the same thing. Theres gonna have a property where the user set to true for "improve cut". But I also add another property to set the quality of this cut geometry improvement, as show in the video and the images below.

https://github.com/user-attachments/assets/5995159a-d35c-4afd-9878-d56e4e2b3343

[image]

This function need improvements yet. Next step is to find a way to smooth the edges before the cut operation in the code.

luzpaz, 2024-11-27:
> Version: 2024.1006.0.14555 (Git shallow) AppImage

An aside, @free777cat where did you install this version of FreeCAD from ?

free777cat, 2024-11-27:
[https://github.com/realthunder/FreeCAD/releases](url)

[image]