shaise/FreeCAD_SheetMetal #402: Error when loading a file with extruded cutout features

shaise, 2024-11-21:
### Describe your issue

Loading a file with extruded cutout features generates some errors, even though the file seems ok after it is loaded.
I was able to remove some of the errors (missing loads() function) but others still exist
To reproduce, just open the attached file.

### FreeCAD version info + SheetMetal WB version

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * fasteners 0.5.31
  * OpenTheme 2024.9.1
  * sheetmetal 0.5.6
  * STEMFIE 0.2.4
```

### Put here any screenshots or videos (optional)

[test-cutout2.zip](https://github.com/user-attachments/files/17845980/test-cutout2.zip)

shaise, 2024-11-21:
@sheetmetalman , I'm still looking for the problem. Please update your code as I fixed some of it.

closed 2024-11-21 by shaise

sheetmetalman, 2024-11-22:
Thank you so much, @shaise!

Yes! I will update my code here same as yours updates! Tanks again!