shaise/FreeCAD_SheetMetal #406: Fold a Wall  -   Works but gives certain messages

free777cat, 2024-11-25:
### Describe your issue

PartDesign::FeaturePython: Link(s) to object(s) 'Sketch' go out of the allowed scope 'Fold'. Instead, the linked object(s) reside within 'N/A'.

### FreeCAD version info + SheetMetal WB version

```shell
OS: Linux Mint 22 (MATE/mate/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 

  * sheetmetal 0.5.8
```

### Put here any screenshots or videos (optional)

[FaW_messages.zip](https://github.com/user-attachments/files/17900717/FaW_messages.zip)

shaise, 2024-11-26:
Thank you. Yes I also encountered that, but I still did not find the problem because the error is reported by freecad and not sheetmetal. Working on it.

free777cat, 2024-11-26:
It might not matter, but the realthunder version doesn't display these notifications.

closed 2025-02-02 by free777cat