shaise/FreeCAD_SheetMetal #408: Unfold (Either I misunderstood something, or this is a level 60 bug)

free777cat, 2024-11-27:
### Describe your issue

error at Unfold 60 degrees

### FreeCAD version info + SheetMetal WB version

```shell
[code]
OS: Linux Mint 22 (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 2024.1006.0.14555 (Git shallow) AppImage
Build type: Release
Branch: (HEAD detached at 20241003stable)
Hash: 8058dcf52c2661edabf36d8147d5b8a187378d2d
Python 3.11.9, Qt 5.15.13, Coin 4.0.1, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 

  * sheetmetal 0.5.10

[/code]
```

### Put here any screenshots or videos (optional)

[bug_60_level.zip](https://github.com/user-attachments/files/17936554/bug_60_level.zip)
[https://youtu.be/Yzg94mp_Uuc](url)

shaise, 2024-11-27:
He He! very strange...

EstebanFPV, 2024-11-27:
looks like level 60 to me

PR #433 (2025-01-06): add new unfolder backend
This PR is to make sure we don't forget to add the new dependency in `package.xml` and close relevant issues. @shaise will plug in the backend functions to the GUI as needed.

This new unfolder:
- Uses the `networkx` python module for analyzing the sheet faces
- Should be indifferent to refined vs. unrefined shapes. Seam lines don't matter in most cases
- Is **5-50x faster than the old unfolder**  

closes #408
closes #361 
closes #345
closes #284 
closes #279 
closes #278 
closes #274 
closes #272 
closes #254 
closes #235 
closes #225 
closes #203 
closes #25 
closes [... cut]
closed 2025-01-06 by shaise