shaise/FreeCAD_SheetMetal #409: Bend - two questions

free777cat, 2024-11-29:
### Describe your issue

[image]
1. must be added gap1, gap2   -  the design cycle will be completed
[image]

2. When you reopen the task panel for bend (double-click on bend ) and close it, a notification is issued:

08:11:57  Traceback (most recent call last):
  File "/home/X/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalCmd.py", line 1860, in unsetEdit
    Gui.Selection.setSelectionStyle(Gui.Selection.SelectionStyle.NormalSelection)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: module 'Selection' has no attribute 'setSelectionStyle'

### FreeCAD version info + SheetMetal WB version

```shell
[code]
OS: Linux Mint 22 (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 2024.1006.0.14555 (Git shallow) AppImage
Build type: Release
Branch: (HEAD detached at 20241003stable)
Hash: 8058dcf52c2661edabf36d8147d5b8a187378d2d
Python 3.11.9, Qt 5.15.13, Coin 4.0.1, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 

  * sheetmetal 0.5.10

[/code]
```

shaise, 2024-11-29:
Side A and Side B are the gaps 1 and 2...
Also other parameters missing. everything in progress now. I'm adding GUI to all other sheetmetal functions

free777cat, 2024-11-29:
I think it is not one and the same

gaps
[image]
[image]

Side A and Side B
[image]
[image]
[image]

free777cat, 2024-11-29:
++ gap1, gap2

shaise, 2024-11-29:
You are correct!
working on it.
Regarding the error message it is only on linkstage version. Its not supporting multi select (or maybe doing it otherwise) Anyway, I will at least suppress this error.

shaise, 2024-11-29:
by the way, do you see the advances parameters as well?

[image]

closed 2024-11-29 by shaise

free777cat, 2024-11-30:
Sorry for the delay in replying. Due to shelling by the Russians, we have outages with electricity and the Internet (they just turned it on).
In my opinion, the idea with tabs is very successful. I will practically check it tomorrow.
[image]

free777cat, 2024-11-30:
Maximum Extend Distance 
reduce to the form
Max Extend Distance

this would reduce the overall width
A slightly narrower menu is easier to work with

shaise, 2024-11-30:
@free777cat , Thank you for helping me with sheetmetal, especially with the condition in the Ukraine.
I have changed the Max Extend as you have suggested.

free777cat, 2024-11-30:
Thank.
Didn't we deserve "Relief Cuts" on the other "rib" "Bend"? ))
[image]

free777cat, 2024-11-30:
Maybe it's difficult, but another "Relief Cuts_2" menu on this tab will look very nice))

shaise, 2024-11-30:
You mean different parameters for relief cut 1 and 2?

free777cat, 2024-11-30:
Yes. 
But it is possible to make       relief cut_2 = relief cut_1        according to the label

free777cat, 2024-11-30:
Fold a wall - menu
[image]
invert   -   add parameter

Flip Direction  -   rename parameter  ( invertbend )

shaise, 2024-11-30:
Thank you. Fixed the title.
Regarding invert bend, I'm using the design as decided by the Design Work Group. https://github.com/shaise/FreeCAD_SheetMetal/issues/324

commit 4cb3bb22 (2024-11-29): Add gap1, gap2 to flange task panel. change to tabs, add perforation gui. Fixes #409