shaise/FreeCAD_SheetMetal #412: Milestones for ver 1.0 of sheetmetal

shaise, 2024-12-04:
### Describe the feature

[] Remove code duplication. Add helper function for common tasks
[] Add GUI (task panel) to all functions
[] Sticky parameter values when a function is reused
[] Parametrize unfolding (using manual update button, because it is slow)
[] Update translation files on crowdin
 

### Would you like to work on this issue?

Yes

shaise, 2024-12-04:
@hasecilu,
I have implemented GUI panels to all missing function, Can you please update Crowdin with these changes?

hasecilu, 2024-12-04:
Hi, file has been updated.

ma00aem, 2025-01-13:
How can i test this release? Is it updated automatically or how to download it?

shaise, 2025-01-13:
Everything marked "V" are already in the latest release.

shaise, 2025-08-25:
out of all the issues, the TNP is the most important. It looks like someone is working on it.
Regarding other milestones, I can try to attend some, but even if I dont, if TNP is fixed, I think we can mark it as V1.0

luzpaz, 2025-08-25:
> Regarding other milestones, I can try to attend some, but even if I dont, if TNP is fixed, I think we can mark it as V1.0

Makes absolute sense :+1:

luzpaz, 2025-08-31:
@alexneufeld any interest in hitting one of these before the 1.0 release?

Huwyler, 2026-11-07:
Regarding "Make the WB TNP-aware?": #638 is a first step for **Bend**. It gives the intermediate tool shapes in `smBend()` deterministic tags, so Bend faces and edges keep their names across recomputes, parameter changes and save/reopen. It includes regression tests. Other features (Extend, Junction, Relief, …) could follow the same pattern if this approach is OK.

luzpaz, 2026-11-11:
> https://github.com/FreeCAD/FreeCAD/issues/18671  

Does #638 honor the above pending PR ?

Huwyler, 2026-11-11:
Sort of, but @shaise and I agree that workbenches should not be in charge of dealing with the toponaming issue, if it can be fixed by Core FreeCAD. That being said, PartDesign does a similar thing (see discussion in the mentioned PR).

So, I think, it can be used as an intermediate fix, that's why I want to keep the branch available and will rebase it to main from time to time. I don't think it will break anything, even if the issues mentioned here are resolved, but I also think we should not have to depend on it. A fix in the core is just more sound and consistent.

luzpaz, 2026-11-16:
> - [ ] Make moving the tip and inserting of sheetmetal features in PD work  

Is this an upstream dependent feature ?

shaise, 2026-11-16:
not sure