shaise/FreeCAD_SheetMetal #413: Oh,oh))      Irrelevant zeros are not automatically removed

free777cat, 2024-12-04:
### Describe your issue

[image]

### FreeCAD version info + SheetMetal WB version

```shell
[code]
OS: Linux Mint 22 (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 2024.1006.0.14555 (Git shallow) AppImage
Build type: Release
Branch: (HEAD detached at 20241003stable)
Hash: 8058dcf52c2661edabf36d8147d5b8a187378d2d
Python 3.11.9, Qt 5.15.13, Coin 4.0.1, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 

  * sheetmetal 0.6.0
```

shaise, 2024-12-04:
I don't think its a sheetmetal issue. 
It does not happen for me in main and not in realthunder:
[image]

[image]

free777cat, 2024-12-04:
Does your program not work like this video?
https://youtu.be/_wfYyHwqoRs

shaise, 2024-12-04:
Yes indeed but this is same for all other freecad parameters. not only sheetmetal. for example:
in part design:
[image]

free777cat, 2024-12-04:
This is not very critical.  But it works in the editor.
https://youtube.com/shorts/C7976aeWk_Y

shaise, 2024-12-04:
Yes, you are correct. But as said, I use Freecad's Edit boxes, so I have no way to change their behavior.

shaise, 2024-12-04:
Don't know...

hasecilu, 2024-12-05:
Is your FreeCAD setting set corrrectly? I set 3 decimals and that's what I get

[image]

shaise, 2024-12-05:
@hasecilu , His issue is not with the number of zeros after the point but rather from the option to type zeros before the number