shaise/FreeCAD_SheetMetal #414: Rililef Cuts        Round

free777cat, 2024-12-04:
### Describe your issue

Doesn't work when depth =< width
It should work at least up to depth = 0.5width
[image]
[image]
[image]

### FreeCAD version info + SheetMetal WB version

```shell
OS: Linux Mint 22 (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 2024.1006.0.14555 (Git shallow) AppImage
Build type: Release
Branch: (HEAD detached at 20241003stable)
Hash: 8058dcf52c2661edabf36d8147d5b8a187378d2d
Python 3.11.9, Qt 5.15.13, Coin 4.0.1, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 

  * sheetmetal 0.6.0
```

shaise, 2024-12-04:
Thank you! Let me check

shaise, 2024-12-04:
Ok. This is fixed

free777cat, 2024-12-04:
Sorry, but not quite)
https://youtu.be/QPCxNd_SzZ8

free777cat, 2024-12-04:
We need an equation sign ))

shaise, 2024-12-04:
Yes, I know that. 
Using an equation sign will cause the side wall to be of length zero, causing an error. I can try to overcome this, but it will complicate the code and I don't think its that important.

shaise, 2024-12-04:
@free777cat, Ok, I took the challenge. It supports a 1/2 ratio, and even smaller...

pierreporte, 2024-12-04:
This is a duplicate of #309. I’m closing my issue because it doesn’t have any discussion.

shaise, 2024-12-04:
@pierreporte , thanks.
@free777cat , can we close the issue now, as it supports any width/height ratio now?

free777cat, 2024-12-05:
Thank you very much!!!   It turned out better than expected.

closed 2024-12-05 by free777cat