shaise/FreeCAD_SheetMetal #415: self counting spin boxes

FC-FBXL5, 2024-12-04:
### Describe your issue

When I klick on the arrows of the spin boxes in the Task panel of tools related to SheetMetal an endless count/countdown starts. Spin boxes related to other workbenches behave normal.

See https://forum.freecad.org/viewtopic.php?p=787933#p787933

### FreeCAD version info + SheetMetal WB version

```shell
OS: macOS 15.1.1
Architecture: arm64
Version: 1.1.0dev.39416 (Git) Conda
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Reinforcement
  * Movie 2024.2.15
  * freecad.gears 1.3.0
  * sheetmetal.backup1729457505.792129 0.5.1 (Disabled)
  * fasteners 0.5.33
  * sheetmetal 0.6.0
  * Curves 0.6.51
  * Assembly3 0.12.2
```

FC-FBXL5, 2024-12-04:
It happens all the time in any task panel in relation to SheetMetal since at least mid October.
I managed to check this behavior on an older intel Mac now and it doesn't happen, neither with the then existing version 0.4.xx nor with now updated version 0.6.0.
I'll be back with some error messages...

shaise, 2024-12-04:
Can you check other external workbenches, and let me know which one works? I can try to look at their code to see the difference.

FC-FBXL5, 2024-12-04:
Now that I tried to generate error messages it started to run normal.
I deinstalled SM and an old backup folder and reinstalled it (again). Now the error is gone.
I'm puzzled now, that it had survived several updates and at least one reinstallation a few weeks ago.

I think this issue can be closed - sorry for the noise.

closed 2024-12-04 by shaise

FC-FBXL5, 2024-12-04:
It hit again - problem is not solved, this time in the Add Junctions Parameters dialog. A section of the error message looks like this:

```
<class 'Part.OCCError'>: 15StdFail_NotDone BRep_API: command not done
Junction: 15StdFail_NotDone BRep_API: command not done
pyException: Traceback (most recent call last):
  File "/Users/kay/Library/Application Support/FreeCAD/Mod/sheetmetal/./SheetMetalJunction.py", line 93, in execute
    s = smJunction(gap=fp.gap.Value,
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/kay/Library/Application Support/FreeCAD/Mod/sheetmetal/./SheetMetalJunction.py", line 52, in smJunction
    filletedface = joinface.makeFillet(gap, joinface.Edges)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
```

shaise, 2024-12-04:
Since the gui panels are interactive, any intermediate change to the spin boxes immediately causes a recompute, so sometimes the report view is flooded with intermediate errors. I do not think this error is correlated to the free-running spinboxes.
Reopening it anyway

FC-FBXL5, 2024-12-04:
The Flange Parameters dialog when using the Angle spin box results in this error:
`7044.67 <App> Document.cpp(3105): Recursive calling of recompute for document Unnamed
7044.85 <App> Document.cpp(3105): Recursive calling of recompute for document Unnamed
7044.89 <App> Document.cpp(3211): Unnamed#Bend002 still touched after recompute
`

shaise, 2024-12-04:
Not happening to me. Can you attach a file?

FC-FBXL5, 2024-12-05:
Other external workbenches I use don't use a Task panel.
A file is noproblem:
[TestSpinBox.FCStd.zip](https://github.com/user-attachments/files/18020872/TestSpinBox.FCStd.zip)

shaise, 2024-12-05:
Strange, this does not happen to me. Can you try to uninstall/reinstall sheetmetal and try again?

shaise, 2024-12-05:
@FC-FBXL5 , you said that other spinboxes works well, can you give me an example of an EXTERNAL workbench where the spinboxes works well? (on task dialogs, not parameters)

FC-FBXL5, 2024-12-05:
In the Rocket workbench I found a spin box for **Fin count**. It sometimes behaves normal and sometimes it starts counting, but it stops after 3 to 4 steps.
In the Ship workbench is a spin box **Number of Speeds** that behaves normal.
And lots of spin boxes in the Maker workbench and none of the checked ones started to count.

FC-FBXL5, 2024-12-05:
After the reinstallation the counting doesn't start always. And it can be stopped with another click outside the spin box. Then this spin box behaves normal until I use an other spin box and return to this one.

shaise, 2024-12-05:
Hi @FC-FBXL5 ,
I have tested Maker workbench and Ship. Both of them use native QT spins, Not freecad ones. The freecad Spins supports entering math expressions - this is a crucial feature.
So unfortunately I'm still puzzled.

FC-FBXL5, 2024-12-05:
I have learnt to live with this issue and if no one else can confirm this it may be better to not spend too much time on it apart from keeping an eye on it. Maybe it resolves automatically while you improve the Sheetmetal tools. Maybe some cross references mutated to cross interference (if that makes sense).

My aim was to improve the wiki pages according to the GUI command model page, but with every new version it becomes more evident that they need a complete rewrite. I redo the examples with the recent version and run into difficulties to finish the examples. I have no clue if it makes sense to report issues for all oddities I observe like coplanar issues with cutouts for interlocking extended walls, or that the BaseBend feature refuses to be put into a PD Body currently.

I'm sorry for being a bit off topic but Github isn't really my cup of tea and I hardly get used to it.

shaise, 2024-12-06:
@FC-FBXL5 , You are one of may favorite FreeCAD contributors, That why I take your notes seriously. 
I followed your posts on the forum as well.
Feel free to post any issue, no mater how small it is, as I'm trying to iron out as many sheet metal issues as I can now.
Not everything is simple to solve specially geometry issues such as co-planar geometry, However 'BaseBend feature refuses to be put into a PD Body' is something I should definitely solve.
I know github is not a convenient platform, but having a thread here per issue makes it easy for me to track.
Feel free to go off-topic here anytime...

shaise, 2024-12-06:
Update: 'BaseBend feature refuses to be put into a PD Body' should be fixed now. Thanks for the catch!

FC-FBXL5, 2024-12-06:
@shaise, thanks for the kind words. 
SheetMetal is my favorite workbench. I use it for my DIY projects creating leather belts, bags, an boxes. It helps a lot to create hole patterns for the stitching.
It is astonishing that you have managed to create tools that work well in Part and PartDesign environments while others often explain that it is impossible due to the different modeling paradigms

shaise, 2024-12-06:
@FC-FBXL5 , what type of Mac do you use?

FC-FBXL5, 2024-12-06:
It is a MacBook Air M3. I had to replace my 12 year old MacBook a while ago.

shaise, 2024-12-06:
I will probably have to get my hand on a used Mac M series sometime, just to debug freecad.