shaise/FreeCAD_SheetMetal #419: Vertex activates AddJunction command

FC-FBXL5, 2024-12-06:
### Describe your issue

The AddJunction command should be activated when an edge is selected but only a selected vertex activates it.

### FreeCAD version info + SheetMetal WB version

```shell
OS: macOS 15.1.1
Architecture: arm64
Version: 1.1.0dev.39416 (Git) Conda
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Reinforcement
  * Movie 2024.2.15
  * freecad.gears 1.3.0
  * fasteners 0.5.33
  * Rocket 4.0.1
  * Ship 2024.11.26
  * sheetmetal 0.6.1
  * MakerWorkbench 1.0.1
  * Curves 0.6.51
  * Assembly3 0.12.2
```

shaise, 2024-12-06:
You are right! Sorry about that. Its fixed now.

FC-FBXL5, 2024-12-07:
Confirmed.