shaise/FreeCAD_SheetMetal #420: AddWall tool cannot retrieve BaseBend.radius form BaseBend object

FC-FBXL5, 2024-12-07:
### Describe your issue

AddWall tool cannot retrieve BaseBend.radius form BaseBend object. 
The value in the Task panel red and there is an error message in the Report view:
```
<Exception> ObjectIdentifier.cpp(1698): Property 'radius' not found in 'BaseBend.radius'
in property binding 'Unnamed#Bend.radius'
Bend: Property 'radius' not found in 'BaseBend.radius'
in property binding 'Unnamed#Bend.radius'
```

### FreeCAD version info + SheetMetal WB version

```shell
OS: macOS 15.1.1
Architecture: arm64
Version: 1.1.0dev.39416 (Git) Conda
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Reinforcement
  * Movie 2024.2.15
  * freecad.gears 1.3.0
  * fasteners 0.5.33
  * Rocket 4.0.1
  * Ship 2024.11.26
  * sheetmetal 0.6.1
  * MakerWorkbench 1.0.1
  * Curves 0.6.51
  * Assembly3 0.12.2
```

shaise, 2024-12-07:
Can you explain how to reproduce? (and perhaps attach a sample file)

FC-FBXL5, 2024-12-08:
Create a new document.
Create a sketch either in a PD Body or a separate one, closed or open. 
Create a BaseBend from the sketch.
Select one or more edges and run the AddWall function.
The Task panel opens and displays the Radius value in red, and the Report view shows the message mentioned above.

shaise, 2024-12-08:
Thanks, That's exactly what I tried but it did not happen to me. Let me dig dipper.

shaise, 2024-12-08:
I just can not reproduce it.
The strange thing is this error is related to expressions. As if there is an expression in the radius property. But this should not happen if it is a new file.
Can you please check if the Radius property in the task panel have an expression?

shaise, 2024-12-08:
I have manage to reproduce this error the following way:
I did what you said, without getting an error.
I have changed the radius of the AddWall to be an expression BaseBend.Radius (with capital R) then saved the file.
The using a text editor I have changed the Radius to radius (lower case)
When I reopened the file I got this error:
[image]

Here is why I did the test. There is a pending Issue of sheetmetal properties not beginning with Capital letters. However Changing it means old files can not be loaded. So I have added a special code that detect old files and migrate to the new names. However, I have no control over expressions, so these need to be manually fixed.

However, it is still puzzling that you get this error without loading an old file or doing any expressions

FC-FBXL5, 2024-12-08:
You are right, the expression is automatically created and all lower case. The expression button is light blue and doesn't change to at the darker blue until radius is changed to Radius (I haven't observed this behavior earlier with other invalid expressions). With a capital R it works like expected.

FC-FBXL5, 2024-12-08:
You have mentioned before that you now store values to reuse them later. Could it be that I had opened an older file and this expression was stored as a new default value?

pierreporte, 2024-12-08:
I have the same problem.

```
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39503 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: da45328f6f9b35cbbdc7167d7faadb6a07634bdb
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * CosmeticThread3D
  * sheetmetal.backup1731333427.8204525 0.5.3 (Disabled)
  * Curves 0.6.51
  * CurvedShapes 1.0.13
  * sheetmetal 0.6.1
```

shaise, 2024-12-08:
Do anyone of you have an Idea why an expression is created at all? with no user intervention?

shaise, 2024-12-08:
OK found it. it happens when autolink feature is enabled. Never knew I had this feature...
I will fix that.

shaise, 2024-12-08:
It should work now. Thanks for the catch. The auto-link feature is rather nice.

FC-FBXL5, 2024-12-08:
It works well for me now.

free777cat, 2024-12-09:
> It should work now. Thanks for the catch. The auto-link feature is rather nice.

Can you tell us more about the "auto-link feature"?

shaise, 2024-12-09:
The auto-link means the next sheetmetal operations inherit the radius of the bends from the parent ones.

free777cat, 2024-12-09:
You say that you did not know about the existence of this function.

I remember you discussing at length exactly how this function should work for many bends on a part.