shaise/FreeCAD_SheetMetal #423: Bug when creating a new part

guillaumemartin59, 2024-12-14:
### Describe your issue

Hi,

When using the English language setting, everything works fine. However, when using the French language setting, I encountered the following bug:

I go to the menu: Sheet Metal > Créer une forme de base ([en]: Sheet Metal > Create a base form).
Sheet Metal displays an "L-form" by default.
However, in the Task Menu, the form is labeled as "Plat" ([en]: Flat).
This is confusing, especially for new users, as they see an "L-form" but read "Plat" ([en]: Flat). This is particularly troubling for my students, as I am a teacher and would like to use Sheet Metal with them.

Could you please fix this bug? It would greatly improve the experience and make Sheet Metal more accessible in an educational setting.

For reference, this bug does not occur when using the English language. (I haven't tested other languages.)

Thank you so much for your incredible work!

[image]

### FreeCAD version info + SheetMetal WB version

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * sheetmetal 0.6.1
```

### Put here any screenshots or videos (optional)

[image]

shaise, 2024-12-14:
@guillaumemartin59,
Thank you for reporting! It should be fixed now, please update and check and if OK please close the issue.
Feel free to report anything you see.

guillaumemartin59, 2024-12-15:
It works fine. 
Thank you so much to have fixed this bug so quickly !
I close the bug.

closed 2024-12-15 by guillaumemartin59