shaise/FreeCAD_SheetMetal #424: Unable to Make Bend

3x380V, 2024-12-15:
### Describe your issue

Steps to reproduce:
- Std_New
- SheetMetal_BaseShape
- select edge, SheetMetal_AddBend
```
  File "SheetMetalBend.py", line 136, in execute
    s = smSolidBend(
  File "SheetMetalBend.py", line 105, in smSolidBend
    return resultSolid
<class 'UnboundLocalError'>: local variable 'resultSolid' referenced before assignment
```
Behaves the very same way on 0.21.2 and current FreeCAD main. SheetMetal is V0.6.02-beta from git.
Just side note:
https://github.com/shaise/FreeCAD_SheetMetal/blob/3b4e2a4109039249e3b5b90126c12bbdb7bbc056/SheetMetalBend.py#L93-L105
Unconditionally returning variable defined inside conditional block does not seem to be good idea despite what underlying issue is.

### FreeCAD version info + SheetMetal WB version

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US) [ OS: English/United States (en_US) ]
Installed mods: 
  * fasteners 0.5.33
  * Assembly4 0.50.16-dev
  * sheetmetal 0.6.2
```

shaise, 2024-12-15:
I do need to fix this code but the error is because you used the wrong tool. you should use the _addWall tool. The addBend is for rounding sharp (non sheetmetal) corners.

3x380V, 2024-12-15:
Gah, indeed. Used version 0.4 last time and didn't noticed icon changes. Even writing this issue did not ring the bell. Closing this one.

closed 2024-12-15 by 3x380V