shaise/FreeCAD_SheetMetal #425: Unfold doesn't work. Error

free777cat, 2024-12-21:
### Describe your issue

Unfold doesn't work. Error

### FreeCAD version info + SheetMetal WB version

```shell
OS: Linux Mint 22.1 (MATE/mate/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 

  * sheetmetal 0.6.10
```

### Put here any screenshots or videos (optional)

08:58:46  pyException: Traceback (most recent call last):
  File "/home/X/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalUnfoldCmd.py", line 179, in execute
    shape, foldComp, norm, _thename, _err_cd, _fSel, _obN = SheetMetalUnfolder.getUnfold(
                                                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/X/.local/share/FreeCAD/Mod/temp/SheetMetalUnfolder.py", line 2956, in getUnfold
    normalVect = subelement.normalAt(0, 0)
                 ^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'str' object has no attribute 'normalAt'
08:58:46  Unfold: 'str' object has no attribute 'normalAt'

shaise, 2024-12-21:
Please add file

free777cat, 2024-12-21:
[Нова тека.zip](https://github.com/user-attachments/files/18217750/default.zip)

shaise, 2024-12-21:
I'm sorry but all the parts are flat. What do you want to unfold?
[image]

free777cat, 2024-12-21:
Sorry
[err.zip](https://github.com/user-attachments/files/18217837/err.zip)

shaise, 2024-12-21:
strange. for me it works (i just loaded your file and recalculate)
[image]
Maybe its a linux issue? I will try to get a linux machine

shaise, 2024-12-21:
its not a linux issue. it seems you do not have the latest code. the line with the error code is from old version of the file.
try to completely uninstall and reinstall sheetmetal

free777cat, 2024-12-21:
* sheetmetal 0.6.11

18:30:39  pyException: Traceback (most recent call last):
  File "/home/qqq/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalUnfoldCmd.py", line 200, in execute
    shape, foldComp, norm, _thename, _err_cd, _fSel, _obN = SheetMetalUnfolder.getUnfold(
                                                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/qqq/.local/share/FreeCAD/Mod/temp/SheetMetalUnfolder.py", line 2956, in getUnfold
    normalVect = subelement.normalAt(0, 0)
                 ^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'str' object has no attribute 'normalAt'
18:30:39  Unfold: 'str' object has no attribute 'normalAt'

free777cat, 2024-12-21:
/sheetmetal/./SheetMetalUnfoldCmd.py
                      |
_____________|

and should this "dot" be there?

shaise, 2024-12-21:
the sheetmetalunfolder.py does not have this line:
[image]

please remove this file and install the new one

shaise, 2024-12-21:
"Mod/temp/SheetMetalUnfolder.py"

remove the temp folder, it should be sheetmetal folder

closed 2024-12-21 by free777cat