shaise/FreeCAD_SheetMetal #426: Automatic unfolding of all parts in the model (SMUnfoldUnattended)

GS90, 2024-12-21:
### Describe the feature

Hi,
I'm not entirely sure that this is a "feature" and not a "bug", since we are talking about functionality that was already available...

Complex parametric models consist of dozens or even hundreds of sheet metal parts.

There is a need for convenient and quick translation of all parts into "blanks" (sheet metal flat patterns) for subsequent manufacturing.

Previously, the workbench had a command "SMUnfoldUnattended" that allowed you to use the unfolding function from a third-party script similar to mine, a simple example - https://github.com/GS90/addFC/issues/3.

Such functionality for processing sheet parts is quite rare in CAD systems, but very much needed in production tasks.

Now with the introduction of parametric unfolding, the "SMUnfoldUnattended" function has disappeared and this is a big blow to the functionality in general and especially to labor productivity...
Scripts for automatic processing of parts have stopped working.
Translating a hundred parts into *.dxf manually is a problem.

I sincerely believe that it is necessary to provide the ability to use the unfolding in automatic mode, and the user interface for the team is not always necessary.

### Additional context

[image]

Here is an example of my model consisting of more than 50 different sheet metal parts, with the function of automatic export to *.dxf, the time for its processing by the method I described above is seconds.

### Would you like to work on this issue?

None

shaise, 2024-12-21:
hi @GS90 ,
Sorry about that, I can reinstate this functionality, can you share the part of your script that initiate this command? specifically, how do you select the faces/objects you want to unfold?

GS90, 2024-12-21:
A very simple algorithm:

1. Create a copy of the object in the active document, using a method similar to "Part SimpleCopy"
2. Search all faces and find the required one with the largest area
3. Select only the found face and call "SMUnfoldUnattendedCommandClass"

```
# reproduction:
shape = Part.getShape(body, '', needSubElement=False, refine=False)
ad.addObject('Part::Feature', 'Reproduction').Shape = shape
body = ad.ActiveObject

# find the largest face:
faces = body.Shape.Faces
target, a, n = [0.0, 0], 0, 0
for f in faces:
    a = round(f.Area, 2)
    n += 1
    if a >= target[0]:
        target[0] = a
        target[1] = n

# selection:
face = 'Face' + str(target[1])
FreeCAD.Gui.Selection.addSelection(ad.Name, body.Name, face, 0, 0, 0)

# k-factor:
FreeCAD.ParamGet(
    'User parameter:BaseApp/Preferences/Mod/SheetMetal').SetFloat(
    'manualKFactor', k_factor)

# unfold:
msg = f'{d}: {material} ({thickness}) {k_factor}\n'
FreeCAD.Console.PrintMessage(msg)
if u is not None:
    u.Activated(None)  # from SheetMetalUnfoldCmd import SMUnfoldUnattendedCommandClass as u
FreeCAD.Gui.Selection.clearSelection()

us = ad.Unfold_Sketch  # ad = FreeCAD.ActiveDocument
```

https://github.com/GS90/addFC/blob/main/addFC_Unfold.py

P.S. A copy in the active document is needed for quick access to the created sketches, and the unfolding function itself works faster with a copy.

shaise, 2024-12-21:
I have reinstated the command. As before, you need to unfold manually at least once so the parameters will be used in the following unattended unfolds.
let me know if it works

GS90, 2024-12-21:
Thank you very much.

At first glance everything seems to work correctly now.

But I'll test it a bit before closing this issue.

shaise, 2024-12-21:
Great. And sorry about that.

closed 2024-12-31 by GS90