shaise/FreeCAD_SheetMetal #427: Peldaño con curva

sanfemar, 2024-12-22:
### Describe your issue

Cuándo creo un Bend ( material interior ) en un peldaño con curva, el Bend se deforma. 
[image]

### Put here any screenshots or videos (optional)

[peldaño.zip](https://github.com/user-attachments/files/18221523/peldano.zip)

shaise, 2024-12-22:
"Material inside" Does not work good with curved edges. This is a limitation. It is complicated to fix.

Google translate:
"Material interior" No funciona bien con bordes curvos. Esto es una limitación. Es complicado de solucionar.

luzpaz, 2025-08-31:
What should we rename this to ?