shaise/FreeCAD_SheetMetal #429: Base Wall, FreeCad linkstage err = thickness 0 or minus,  radius= minus

free777cat, 2024-12-26:
### Describe your issue

https://youtu.be/LEjgnitkAwI

Version: 1.1.0dev.39741 (Git) Conda AppImage also gives an error

A mistakenly inserted negative value is remembered and used in the next operation even in a new file (this can be very confusing)

### FreeCAD version info + SheetMetal WB version

```shell
[code]
OS: Linux Mint 22.1 (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 2024.1006.0.14555 (Git shallow) AppImage
Build type: Release
Branch: (HEAD detached at 20241003stable)
Hash: 8058dcf52c2661edabf36d8147d5b8a187378d2d
Python 3.11.9, Qt 5.15.13, Coin 4.0.1, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 
  
  * sheetmetal 0.6.13
 
[/code]
```

shaise, 2024-12-26:
is it only on linkstage?

free777cat, 2024-12-26:
Version: 1.1.0dev.39741 (Git) Conda AppImage also gives an error

shaise, 2024-12-26:
Should be ok now

free777cat, 2024-12-27:
It already works.

And is it very difficult to create an algorithm for changing values ​​that would work like this: 3>2>1>0.9>0.8>.....0.1>0.09>0.08....0.01>0.009>0.008..... ?
For such cases, it would be very obvious and logical.

shaise, 2024-12-27:
You mean when you press the arrow keys?

free777cat, 2024-12-27:
Yes

free777cat, 2024-12-27:
Or scroll the mouse wheel

shaise, 2024-12-27:
it is not supported by QT, but I agree it's a good idea. Let me see what I
can do.

shaise, 2024-12-28:
Unfortunately the input field is internal to freecad so I can not change this behavior without changing freecad core code